WebODM/WebODM #1252: Experiencing many issues with duplicating Tasks and permissions errors, etc

TimeScience, 2022-10-18:
Errors:  _ff12f6b2-1d6a-412a-8d4f-8b1600e066e9 not found_
EACCES: permission denied, mkdir 'data/3247abc3-814b-4593-9efd-f38ceb479528'
fb0e6257-2446-4719-a0ca-a693240230c9: _" Processing node went offline. This could be due to insufficient memory or a network error."_
[image]

pierotofy, 2022-10-18:
Hi @TimeScience,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-10-19 19:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

TimeScience, 2022-10-18:
Actually, never mind, this was an older run of the current dataset I'm having issues with

closed 2022-10-18 by TimeScience