WebODM/WebODM #1259: multiple configurations found for submodule.nodeodm (...)

merkato, 2022-10-26:
### How did you install WebODM (docker, installer, etc.)?

docker on Ubuntu 22.04

### What is the problem?

on git stage of updating WebODM with ./webodm.sh update this warning popups:
`
Updating WebODM...
git pull origin master
(...)
Resolving deltas: 100% (16488/16488), completed with 352 local objects. From https://github.com/OpenDroneMap/WebODM
• branch master -> FETCH HEAD
ed55a9d3..de65e1lb master
origin/master
warning: 80005bb79b4ce4e7769387824f71e91992d7ce80:.gitmodules, multiple configurations found for 'submodule.nodeodm/external/NodeODM.path. Skipping second one!
warning: 80005bb79b4ce4e7769387824f71e91992d7ce80:.gitmodules, multiple configurations found for submodule.nodeodm/external/NodeODM.url. Skipping second one!
Updating ed55a9d3..de65ellb
Fast-forward
`

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include 

On 25th of October after ./webodm.sh update problem arise, next day is not reproducible.

pierotofy, 2022-10-26:
Hi @merkato,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-10-27 01:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

pierotofy, 2022-10-28:
Thanks for the report; it's just a warning, you can safely ignore it. :+1:

closed 2022-10-28 by pierotofy