WebODM/WebODM #1263: Proposal: some modifications to run on Podman system

Majstor, 2022-11-08:
Hi,

Firstly i have to say i have reached to run on a pure podman's system like Fedora, but i'm not really in ease with systems container concept in general (Docker, Podman or wathever).

I had have to modify several files. I'll try explain why.

1. webodm.sh. I replaced all "docker" occurences by podman (like "podman-compose"), and maybe others things (like changing escaping quotes), but i don't think it's necessary because Fedora proposes a package who gives some docker's aliases (typing in prompt "docker-compose" is a an alias to "podman-compose"). I'm not sure. See my file about.
2. docker-compose.yml. Need to add a "command" entry and podman seems to don't like some type of escaping (see discuss here : https://community.opendronemap.org/t/installation-on-fedora-31/3302/14? )
3. docker-compose.yml again. I don't know why podman needs to have the local IP address of the machine host (not the "localhost"), but there's a problem to transmit the info by environnment variable (WO_BROKER). I tried to start like this "./webodm.sh --broker redis://192.168.1.55 start" and scripts try against to connect to redis at localhost address. So i modify the docker-compose.yml to insert a value to WO_BROKER (need to begin by "redis://") every occurence to be sure. See my file about.
4. docker-compose.yml + docker-compose.nodeodm.yml Podman seems to give a random "output host" port so i insert value every time it's necessary ("6379:6379" for db postgres, "3000:3000" for NodeODM).

After that, i can run without problem whole WebODM but i have to do more tests seeing if there's a glitch or bug with podman.
Just to remember to can run local NodeODM to use the host machine IP (and not use "localhost:3000").

I don't know if it's possible to homogeinize files to run independantly in Docker nor Podman. I haven't touch other yml files because i don't have possibilities to do the test.

My files :
[docker-compose.nodeodm.yml.txt](https://github.com/OpenDroneMap/WebODM/files/9958685/docker-compose.nodeodm.yml.txt)
[docker-compose.yml.txt](https://github.com/OpenDroneMap/WebODM/files/9958686/docker-compose.yml.txt)
[webodm.sh.txt](https://github.com/OpenDroneMap/WebODM/files/9958687/webodm.sh.txt)

pierotofy, 2022-11-08:
Hi @Majstor,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2022-11-09 02:30) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Majstor, 2022-11-08:
Other informations : as i said here : https://github.com/OpenDroneMap/WebODM/issues/1161
Before, i had to modify only "docker-compose.yml" to run WebODM (so i haven't accused you Piero...) before Podman authors had changed to "v4 API". Podman has deep changes with the v4 API. I think it's a necessary to discuss about it.

pierotofy, 2022-12-14:
I wrote a proof of concept that can indeed use podman to run WebODM. https://github.com/pierotofy/WebODM/tree/podman

It doesn't seem a painful transition, just minor tweaks; I have concerns about supporting two containerization systems, (handling the quirks of one is bad enough) and long term I think podman could be the way to go, mostly because it's FOSS and actually seems to be more performant on Mac (and there Docker Desktop is just terrible).

I'm not sure how a smooth transition could be planned, but it's something that needs to be done as it's super important not to break existing users' installations.

Majstor, 2023-04-28:
Hi Piero,

Sorry for late answer.
Your webodm.sh proposal is just perfect. Only as i said there's problem with environment variables and their transmissions.

michaelquickct, 2023-09-29:
Bumping this thread. I'm a Red Hat Solutions Architect aligned to federal science organizations, who also does quite a bit of WebODM based mapping on the side. Is there any interest in planning a migration path to Podman or is this still backburner territory? Happy to assist in planning and code/docs contrib if there is a real desire. I'd prefer to run rootless.

Saijin-Naib, 2023-09-29:
Welcome! That sounds amazing. Do you have any experience with trying WebODM rootless in PodMan already?

pierotofy, 2023-09-30:
Hi @michaelquickct :wave: It's certainly something that we'd like to do and we'd welcome contributions to implement it (and maintain it). The number one priority for the project at the moment is to not break the existing scripts/workflows with docker, because we have many users that use/will continue to use docker for quite a while, but support for podman as another containerization provider would be most welcome, with the goal of making it the default at some point in the future.