WebODM/WebODM #1264: Duplicating task doubles number of images in Task

TimeScience, 2022-11-11:
### How did you install WebODM (docker, installer, etc.)?

Australian Scalable Drone Cloud fork

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 107 on Windows 10

### What is the problem?

Duplicating a tasks causes the number of images in the task to be doubled, or at least ODM reports this. not clear if it actually is doubled or just a bug in image counting upstream.
See image... The first (failed) task is a duplicate of the task below it.
Note that I can't get the output log file to open, presumably because it is so long, otherwise I'd upload it as well. If there is a standard URL to directly download the log file itself, let me know and I'll pull the log for you.

[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

@pierotofy, I've shared the project with this issue with you (I've seen this issue a bunch of times but this is the most recent issue)
Let me know if you need to be shared read/write rather than read-only

Saijin-Naib, 2022-11-12:
Can you replicate this in the upstream code? I've not been able to locally with the default node.

Does this occur against all nodes you have in your instance?

TimeScience, 2022-11-24:
> Can you replicate this in the upstream code? I've not been able to locally with the default node.
> 
> Does this occur against all nodes you have in your instance?

@OKaluza, can you weigh in on this re reproducibility?

pierotofy, 2022-12-14:
Closing for now, if you have reproduction steps for a vanilla installation of WebODM feel free to re-open. :pray:

closed 2022-12-14 by pierotofy

OKaluza, 2023-01-05:
@TimeScience I think I've fixed this in our case, I doubt it will be an issue for most other ODM users, but leaving a few notes here in case it helps anyone in the future.
- It can happen with a new task or imported task too, not just duplicating a task.
- Duplicates of each image are appearing as the image file with an underscore appended, eg: IMG1234.JPG and IMG1234_.JPG
- The database and disk store in WebODM has the correct image set, but duplicates appear in processing : after processing the images.json contains both the originals and the duplicates.
- This function appears to be where the duplicates are being detected and the underscore added https://github.com/OpenDroneMap/ClusterODM/blob/master/libs/taskNew.js#L34 and https://github.com/OpenDroneMap/NodeODM/blob/master/libs/taskNew.js#L52
- The way our ClusterODM and NodeODM instances were set up led to them sharing their "tmp" directory (a symlink to our large scratch volume), once I fixed this so ClusterODM and each of the NodeODM instances had their own distinct tmp paths the problem went away.
- I stopped digging at this point, because I've wasted enough time on it but guessing somehow files saved in the tmp dir by one node were being found by another and it ended up bundling them all together, once I was able to reliably duplicate the issue it was happening about 50% of the time for a given dataset. I wouldn't have thought this possible due to the UUID subdir for each task, but anyway - just happy it seems to be fixed now!