WebODM/WebODM #1269: WebODM does not process new projects after update.

dinoguys-R-worthless, 2022-11-30:
### How did you install WebODM (docker, installer, etc.)?

Windows installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 108, Windows 10

### What is the problem?

I updated to the new version yesterday and now nothing will run. When I load images it gets to the "Resizing Images" progress bar and never gets past it. The gear symbol turns but the progress bar doesn't fill. I've tried it with sets as small as 40 photos and it still doesn't go. The settings are all on Default for the project.

pierotofy, 2022-11-30:
If you go to `Tools` --> `View Logs`, after the task gets stuck, is there any error messages in the `app.log` file?

dinoguys-R-worthless, 2022-11-30:
> 

I'll check really quickly.

dinoguys-R-worthless, 2022-11-30:
Here is what is has:

"Booting WebODM 1.9.16 build 74..."
"Trying to connect to NodeODM localhost:29301..."
"Attempt 0 failed, retrying..."
"nodeodm: info: Authentication using SimpleToken\r\n"
"nodeodm: info: Listening on 0.0.0.0:6367 UDP for progress updates\r\n"
"nodeodm: info: No tasks dump found\r\n"
"nodeodm: info: Checking for orphaned directories to be removed...\r\n"
"nodeodm: info: Server has started on port 29301\r\n"
"Trying to connect to NodeODM localhost:29301..."
"Connected to NodeODM 2.2.0"
"redis-server:                 _._                                                  \n           _.-``__ ''-._                                             \n      _.-``    `.  `_.  ''-._           Redis 3.2.100 (00000000/0) 64 bit\n  .-`` .-```.  ```\\/    _.,_ ''-._                                   \n (    '      ,       .-`  | `,    )     Running in standalone mode\n |`-._`-...-` __...-.``-._|'` _.-'|     Port: 29379\n |    `-._   `._    /     _.-'    |     PID: 6404\n  `-._    `-._  `-./  _.-'    _.-'                                   \n |`-._`-._    `-.__.-'    _.-'_.-'|                                  \n |    `-._`-._        _.-'_.-'    |           http://redis.io        \n  `-._    `-._`-.__.-'_.-'    _.-'                                   \n |`-._`-._    `-.__.-'    _.-'_.-'|                                  \n |    `-._`-._        _.-'_.-'    |                                  \n  `-._    `-._`-.__.-'_.-'    _.-'                                   \n      `-._    `-.__.-'    _.-'                                       \n          `-._        _.-'                                           \n              `-.__.-'                                               \n\n[6404] 30 Nov 12:02:53.899 # Server started, Redis version 3.2.100\n"
"redis-server: [6404] 30 Nov 12:02:54.074 * DB loaded from disk: 0.174 seconds\n[6404] 30 Nov 12:02:54.074 * The server is now ready to accept connections on port 29379\n"
"Trying to connect to postgres localhost:29543..."
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/pg_ctl: server starting\r\n"
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/psql: psql: FATAL:  the database system is starting up\r\n"
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/psql: exited with code 2"
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/psql: psql: FATAL:  the database system is starting up\r\n"
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/psql: exited with code 2"
"Attempt 0 failed, retrying..."
"Trying to connect to postgres localhost:29543..."
"D:\\WebODM\\resources\\app\\apps\\pgsql/bin/psql: exited with code 0"
"python: Operations to perform:\r\n  Apply all migrations: admin, app, auth, contenttypes, guardian, nodeodm, sessions\r\nRunning migrations:\r\n  No migrations to apply.\r\n"
"python: exited with code 0"
"python: exited with code 0"
"python: INFO:waitress:Serving on http://127.0.0.1:29800\r\n"
"python: [2022-11-30 12:03:22,371: INFO/MainProcess] Connected to redis://:**@127.0.0.1:29379//\r\n"
"python: [2022-11-30 12:03:22,377: INFO/MainProcess] mingle: searching for neighbors\r\n"
"python: [2022-11-30 12:03:23,453: INFO/MainProcess] mingle: all alone\r\n"
"python: [2022-11-30 12:03:23,551: INFO/MainProcess] celery@LAPTOP-22ED2B0F ready.\r\n"
"python: [2022-11-30 12:03:23,557: INFO/MainProcess] pidbox: Connected to redis://:**@127.0.0.1:29379//.\r\n"
"python: [2022-11-30 12:03:23,705: INFO/MainProcess] Received task: worker.tasks.export_raster[b20d5539-cf91-4e51-95d0-4a4ebed4deb5]  \r\n"
"python: INFO Exporting raster D:\\WebODM\\resources\\app\\data\\media\\project/22/task/fa7100d7-7624-49f9-816b-7a357ff5e7c4/assets\\odm_orthophoto\\odm_orthophoto.tif with options: {\"epsg\": 32612, \"expression\": null, \"format\": \"kmz\", \"rescale\": null, \"color_map\": null, \"hillshade\": null, \"asset_type\": \"orthophoto\", \"name\": \"County Road 8 - 5/9/2015\"}\r\n[2022-11-30 12:03:23,705: INFO/MainProcess] worker.tasks.export_raster[b20d5539-cf91-4e51-95d0-4a4ebed4deb5]: Exporting raster D:\\WebODM\\resources\\app\\data\\media\\project/22/task/fa7100d7-7624-49f9-816b-7a357ff5e7c4/assets\\odm_orthophoto\\odm_orthophoto.tif with options: {\"epsg\": 32612, \"expression\": null, \"format\": \"kmz\", \"rescale\": null, \"color_map\": null, \"hillshade\": null, \"asset_type\": \"orthophoto\", \"name\": \"County Road 8 - 5/9/2015\"}\r\n"
"python: D:\\WebODM\\resources\\app\\apps\\grass78\\Python39\\lib\\site-packages\\rio_tiler\\reader.py:385: AlphaBandWarning: Alpha band was removed from the output data array\r\n  warnings.warn(\r\n"
"python: INFO Booting WebODM 1.9.16\r\nINFO:app.logger:Booting WebODM 1.9.16\r\n"
"python: INFO Running npm install for [coreplugins.changedetection.plugin]\r\nINFO:app.logger:Running npm install for [coreplugins.changedetection.plugin]\r\n"
"python: WARNING npm is not installed, will skip this plugin\r\nWARNING:app.logger:npm is not installed, will skip this plugin\r\n"
"python: INFO Running npm install for [coreplugins.cloudimport.plugin]\r\nINFO:app.logger:Running npm install for [coreplugins.cloudimport.plugin]\r\n"
"python: WARNING npm is not installed, will skip this plugin\r\nWARNING:app.logger:npm is not installed, will skip this plugin\r\n"
"python: INFO Running npm install for [coreplugins.dronedb.plugin]\r\nINFO:app.logger:Running npm install for [coreplugins.dronedb.plugin]\r\n"
"python: WARNING npm is not installed, will skip this plugin\r\nWARNING:app.logger:npm is not installed, will skip this plugin\r\n"
[... cut ...]
"python: INFO Registered [coreplugins.diagnostic.plugin]\r\nINFO:app.logger:Registered [coreplugins.diagnostic.plugin]\r\nINFO Registered [coreplugins.fullscreen.plugin]\r\nINFO:app.logger:Registered [coreplugins.fullscreen.plugin]\r\nINFO Registered [coreplugins.lightning.plugin]\r\nINFO:app.logger:Registered [coreplugins.lightning.plugin]\r\n"
"python: INFO Registered [coreplugins.measure.plugin]\r\nINFO:app.logger:Registered [coreplugins.measure.plugin]\r\nINFO Registered [coreplugins.posm-gcpi.plugin]\r\nINFO:app.logger:Registered [coreplugins.posm-gcpi.plugin]\r\n"

dinoguys-R-worthless, 2022-11-30:
That is with 30 images. Looks like I'm missing some Python stuff. I'm not a Python wizard. Should I uninstall and reinstall WebODM?

pierotofy, 2022-11-30:
A reinstall would be a good thing to try! Nothing seems out of the ordinary from the log.

dinoguys-R-worthless, 2022-11-30:
> 

I still have the WebODM manager from before it went Windows native. Should I get rid of that?

dinoguys-R-worthless, 2022-11-30:
But I need to keep Docker?

ryanflahertyOG, 2022-11-30:
I'm also unable to successfully process new projects since the update. I successfully ran a project yesterday prior to the update with the same set of images. Now I'm getting the following error in the log:
[INFO]    running "C:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\OpenMVS\DensifyPointCloud" "C:\WebODM\resources\app\apps\NodeODM\data\2a3fd37a-1675-4c0f-88fb-8dc680eb3b17\opensfm\undistorted\openmvs\scene.mvs"  --resolution-level 3 --dense-config-file "C:\WebODM\resources\app\apps\NodeODM\data\2a3fd37a-1675-4c0f-88fb-8dc680eb3b17\opensfm\undistorted\openmvs\Densify.ini" --max-resolution 5464 --max-threads 16 --number-views-fuse 2 --sub-resolution-levels 2 -w "C:\WebODM\resources\app\apps\NodeODM\data\2a3fd37a-1675-4c0f-88fb-8dc680eb3b17\opensfm\undistorted\openmvs\depthmaps" -v 0 --cuda-device -2
===== Dumping Info for Geeks (developers need this to fix bugs) =====
Child returned 1
Traceback (most recent call last):
File "C:\WebODM\resources\app\apps\ODM\stages\odm_app.py", line 81, in execute
self.first_stage.run()
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 382, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 382, in run
self.next_stage.run(outputs)
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 382, in run
self.next_stage.run(outputs)
[Previous line repeated 1 more time]
File "C:\WebODM\resources\app\apps\ODM\opendm\types.py", line 361, in run
self.process(self.args, outputs)
File "C:\WebODM\resources\app\apps\ODM\stages\openmvs.py", line 117, in process
raise e
File "C:\WebODM\resources\app\apps\ODM\stages\openmvs.py", line 103, in process
run_densify()
File "C:\WebODM\resources\app\apps\ODM\stages\openmvs.py", line 99, in run_densify
system.run('"%s" "%s" %s' % (context.omvs_densify_path,
File "C:\WebODM\resources\app\apps\ODM\opendm\system.py", line 109, in run
raise SubprocessException("Child returned {}".format(retcode), retcode)
opendm.system.SubprocessException: Child returned 1

===== Done, human-readable information to follow... =====

[ERROR]   Uh oh! Processing stopped because of strange values in the reconstruction. This is often a sign that the input data has some issues or the software cannot deal with it. Have you followed best practices for data acquisition? See https://docs.opendronemap.org/flying/

pierotofy, 2022-11-30:
> But I need to keep Docker?

Nop!

pierotofy, 2022-11-30:
> I'm also unable to successfully process new projects since the update.

Can you share a link to your input images?

pierotofy, 2022-11-30:
1. Make a copy of `C:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\OpenMVS\DensifyPointCloud.exe` 
2. Replace it with this application (from the previous 2.9.2 release) https://drive.google.com/file/d/1_OHRCMMapTEJERs_OHdC2wCh8PPeOLNc/view?usp=share_link
3. Try to rerun the dataset. Does it work?

ryanflahertyOG, 2022-11-30:
> Also something to try:
> 
>     1. Make a copy of `C:\WebODM\resources\app\apps\ODM\SuperBuild\install\bin\OpenMVS\DensifyPointCloud.exe`
> 
>     2. Replace it with this application (from the previous 2.9.2 release) https://drive.google.com/file/d/1_OHRCMMapTEJERs_OHdC2wCh8PPeOLNc/view?usp=share_link
> 
>     3. Try to rerun the dataset. Does it work?

This did the trick!

dinoguys-R-worthless, 2022-11-30:
I uninstalled and reinstalled it. Ran a 25 image test and that went well. I'm running a larger model now and it seems to be going well. However, it did not get the newest version when I reinstalled it (reinstall from old installation file). I'm really excited for ODM 3.0 but maybe I should wait a little while before upgrading? What do you think?

pierotofy, 2022-12-01:
> This did the trick! 

Interesting. I wonder if it's some sort of race condition, it processed fine on my machine (although I don't have 16 cores).

I'm about to release ODM `3.0.1` which will revert some of the changes that *might* be at fault.

dinoguys-R-worthless, 2022-12-01:
> > This did the trick!
> 
> Interesting. I wonder if it's some sort of race condition, it processed fine on my machine (although I don't have 16 cores).
> 
> I'm about to release ODM `3.0.1` which will revert some of the changes that _might_ be at fault and _might_ fix this.

Looks like it isn't GPU processing though. Last time I ran this model it took just under 5 hours (2189 photos). I'm 6:40 in right now and it's only about 20% done. I'll see if I can make a google drive for the photos if you want to give it a shot.

pierotofy, 2022-12-02:
Ok, 3.0.1 is out and should fix this. Try to update and reprocess. If the problem persists, feel free to reopen? :pray:

closed 2022-12-02 by pierotofy

Jarrodwp1, 2023-03-14:
I'm having a similar issue. I have WebODM on Windows 11 from the installer. I get data cannot process when working on a new project. Never had any issues before. I tried uninstalling and reinstalling, but so far, no fix.