WebODM/WebODM #1270: Volume - measure plugin: pile_height_above_dsm raster is not masked (and other issues) before volume computation.

gabriel-de-luca, 2022-12-06:
### How did you install WebODM?

Cloned this repository and run the start bash script.

### What's your browser and operating system? 

Firefox and debian buster.

### What is the problem?

_pile_height_above_dsm_ raster is not masked by _polygon_area_ vector (it is used to adjust the region, but the region is a bounding box) before volume computation.

In addition, differences are observed between the original dsm and the one interpolated with _v.surf.rst_, some of them from the algorithm and others from the chosen area.

### How can we reproduce this? 

Perform a measurement in WebODM. Export _dsm_below_pile_ and _pile_height_above_dsm_ rasters to analyze them.

---

I processed a basic task over the Brighton Beach dataset and performed a volume measurement of a digitized area so that the computation can be analyzed. The points cross irregularities in the DSM and the shape of the area is slightly concave. The volume measures 64.88 cubic meters: 

[image]

Also, included two lines in the _/webodm/coreplugins/measure/calc_volume.py_ file of the worker container, to export the rasters to the root directory in order to analyze them:

```
    # Decimate DSM and generate interpolation of new terrain
    Module("v.surf.rst", input="polygon_points", zcolumn="height", elevation="dsm_below_pile", overwrite=True)
    Module("r.out.gdal", input="dsm_below_pile", output="/dsm_below_pile.tif", overwrite=True)

    # Compute difference between dsm and new dsm
    Module("r.mapcalc", expression='pile_height_above_dsm=dsm-dsm_below_pile', overwrite=True)

    # Set region to polygon area to calculate volume
    Module("g.region", vector="polygon_area")
    Module("r.out.gdal", input="pile_height_above_dsm", output="/pile_height_above_dsm.tif", overwrite=True)
```

(I continue the analysis in comments below in order to save the issue, because I had already written this but a bad click made me lose it).

gabriel-de-luca, 2022-12-06:
Next, opened the original _DSM_, both computed rasters and the _measurement.geojson_ polygon into QGIS. 
Extracted the vertices of the polygon and labled them with Z values from original _DSM_ and _dsm_below_pile_. 
Also, stylized the _pile_height_above_dsm_ raster with a gradient color ramp to see negative values (indicates the _dsm_below_pile_ is over the _DSM_): 

[image]

Conclusions:

- The _pile_height_above_dsm_ raster is not masked with the _polygon_area_ vector, so pixels outside the digitized area are used to compute the volume.
- The Z value in the dsm_below_pile shows differences of up to 0.10 meter from the DSM (_smooth=0_ parameter in _v.surf.rst_ algorithm seems to be needed to avoid that).
- Perhaps the greatest influence on the volumetric computation is due to the pixels that resulted with negative Z in the _pile_height_above_dsm_ raster. I guess the choice of points on irregularities in the terrain produces that, which I consider difficult to solve in an unsupervised way. The calculation of volumes with the raster surface volume algorithm of QGIs returns the following:

Count only volume above zero level: {'AREA': 97.19752889241224, 'PIXEL_COUNT': 27847, 'VOLUME': 102.80085979753616}.

Count only volume below zero level: {'AREA': 59.91642837530608, 'PIXEL_COUNT': 17166, 'VOLUME': -37.924360463088775}.

Subtract volume below zero level: {'AREA': 157.11395726771832, 'PIXEL_COUNT': 45013, 'VOLUME': 64.87649933444739}.

pierotofy, 2022-12-06:
Thanks for looking into this @gabriel-de-luca :pray: and for the detailed analysis.

I think you might be onto something with the `v.surf.rst` observation. I think you're correct in that we should probably pass `smooth=0`.

> pile_height_above_dsm raster is not masked by polygon_area vector (it is used to adjust the region, but the region is a bounding box) before volume computation.

If I remember correctly (it's been a while since I wrote this script), the `g.region` was used solely to speed up computations. The `r.volume` command *should* take into consideration `MASK` when computing results, which is set via `r.mask` (which we call on line 44). https://grass.osgeo.org/grass82/manuals/r.volume.html (since we don't specify the clump parameter: "If a clump map is not specified, the current MASK is used.").

It would be really nice to have a ground truth DSM from which there's known stockpile volume measurements and related polygons from which to run tests against, not sure if you have one that you can share?

pierotofy, 2022-12-06:
p.s. note the `calc_volume.py` is also a GRASS GIS compliant script, which one can open/run directly into GRASS. Not sure if this can help future analysis efforts.

gabriel-de-luca, 2022-12-06:
> The `r.volume` command _should_ take into consideration `MASK` when computing results, which is set via `r.mask` (which we call on line 44)

Yes, it does. But the mask used is the _region_ vector, wich is a buffer of _polygon_area_ vector, used as mask for the interpolated reference surface, and it works.

I think the mask can be updated before the volume computation as:
```
    # Set region and mask to polygon area to calculate volume
    Module("g.region", vector="polygon_area")
    Module("r.mask", vector="polygon_area")

    # Volume output from difference
    Module("r.volume", input="pile_height_above_dsm", f=True)
```

pierotofy, 2022-12-06:
> but I can share DSMs and volume computations

These would be most helpful.

I've tried to use synthetic DSMs to validate the proposed changes (https://github.com/OpenDroneMap/syntheticDSM), but I couldn't verify any meaningful improvement (I suspect because of the perfection of the synthetic data). The changes make sense, I will go ahead and merge #1272 anyway. I will also leave this open for further discussion/verification.

gabriel-de-luca, 2022-12-16:
**Tested the fix and it solves the mask and smooth issues.**

----

> These would be most helpful.  

I prepared four DSM belonging to the same survey. Each zip file contains a dsm, the vector area used to interpolate, the reference surface (manually extracted with interpolations and filtering within the area), the difference between both (dsm - reference) surfaces and the QGIS volume report: 
[test_1.zip](https://github.com/OpenDroneMap/WebODM/files/10242778/test_1.zip)
[test_2.zip](https://github.com/OpenDroneMap/WebODM/files/10242782/test_2.zip)
[test_3.zip](https://github.com/OpenDroneMap/WebODM/files/10242785/test_3.zip)
[test_4.zip](https://github.com/OpenDroneMap/WebODM/files/10242786/test_4.zip)

The only normal case is 1, on which I did the process in GRASS GIS simulating the process that WebODM does.
QGIS report says:

```
Analyzed file: difference_1.tif (band 1)
Volume: 2070.702271899134
Pixel count: 381094
Area: 952.7350000000002 m²
```

And GRASS report says: 
```
Volume report on data from <pile_height_above_dsm@tests> using clumps on <MASK> raster map
Category   Average   Data   # Cells        Centroid             Total
Number     in clump  Total  in clump   Easting     Northing     Volume
-----------------------------------------------------------------------------
       1      2.15    832167  386616   500071.25      130.27          2080.74
-----------------------------------------------------------------------------
                                                Total Volume =        2080.74
```

Difference is around 0.5 %.

Likewise, visually the tension of 40 (default) seems a bit exaggerated to me. Even GRASS warns about an overshoot and the suggestion to increase the tension.  

---
`tension = 40` : 

[image]
---
`tension = 100` : 

[image]
---

The GRASS report for tension 100 says:

```
Volume report on data from <pile_height_above_dsm_100@tests> using clumps on <MASK> raster map
Category   Average   Data   # Cells        Centroid             Total
Number     in clump  Total  in clump   Easting     Northing     Volume
-----------------------------------------------------------------------------
       1      2.14    826709  386616   500071.25      130.27          2067.09
-----------------------------------------------------------------------------
                                                Total Volume =        2067.09
```

The value is a little closer (difference of 0.15 %). But in all cases, my reference surface is subjective and any comparison with an automatic method of producing it must take that into account.

---

I haven't analyzed the other test datasets in GRASS yet, but they are not simple cases. Test 2 in a small sector, but it has a well in the center of the pile, with a lower altitude than those in the perimeter. Tests 3 and 4 are the most common cases, of stacking close to a retaining wall. Test 3 being a little smaller but with a border of highly variable altitudes, and test 4 the largest but a little more continuous.  
I think they can be useful to analyze the possibility of introducing other features, such as 3D points (instead of taking the Z from the DSM), or perform an algebraic operation that prevents the reference surface from being above the DSM. For the current functionality the current script seems correct to me.