WebODM/WebODM #1280: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE

chn-lee-yumi, 2022-12-17:
### How did you install WebODM (docker, installer, etc.)?

```shell
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1
cd WebODM
./webodm.sh start --gpu
```

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 103 on Linux

Ubuntu 22.04

### What is the problem?

Error when running opensfm:

```text
[INFO]    running /code/SuperBuild/install/bin/opensfm/bin/opensfm detect_features "/var/www/data/9331cba0-1b2b-424b-86b1-1a8ba5098d56/opensfm"
2022-12-17 06:55:58,037 INFO: Capping memory usage to ~ 6109.87109375 MB
2022-12-17 06:55:58,037 INFO: Expecting to process 133 images.
2022-12-17 06:55:58,043 INFO: Reading data for image DJI_0659.JPG (queue-size=0
2022-12-17 06:55:58,043 INFO: Reading data for image DJI_0583.JPG (queue-size=0
2022-12-17 06:55:58,043 INFO: Reading data for image DJI_0508.JPG (queue-size=0
2022-12-17 06:55:58,043 INFO: Reading data for image DJI_0430.JPG (queue-size=0
2022-12-17 06:55:58,383 INFO: Reading data for image DJI_0650.JPG (queue-size=1
2022-12-17 06:55:58,384 INFO: Extracting ROOT_SIFT_GPU features for image DJI_0659.JPG
2022-12-17 06:55:58,597 INFO: Reading data for image DJI_0509.JPG (queue-size=1
2022-12-17 06:55:58,598 INFO: Reading data for image DJI_0582.JPG (queue-size=2
2022-12-17 06:55:58,598 INFO: Extracting ROOT_SIFT_GPU features for image DJI_0508.JPG
2022-12-17 06:55:58,598 INFO: Extracting ROOT_SIFT_GPU features for image DJI_0583.JPG
2022-12-17 06:55:58,601 INFO: Reading data for image DJI_0429.JPG (queue-size=1
2022-12-17 06:55:58,603 INFO: Extracting ROOT_SIFT_GPU features for image DJI_0430.JPG
2022-12-17 06:55:58,652 ERROR: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE

Build on <pyopencl.Device 'NVIDIA GeForce GTX 950M' on 'NVIDIA CUDA' at 0x196c900>:

(options: -D WORKGROUP_SIZE=1024 -I /usr/local/lib/python3.8/dist-packages/pyopencl/cl)
(source saved as /tmp/tmphdk6to70.cl)
2022-12-17 06:55:58,653 ERROR: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE

Build on <pyopencl.Device 'NVIDIA GeForce GTX 950M' on 'NVIDIA CUDA' at 0x196c900>:

(options: -D WORKGROUP_SIZE=1024 -I /usr/local/lib/python3.8/dist-packages/pyopencl/cl)
(source saved as /tmp/tmp0qgkmq93.cl)
2022-12-17 06:55:58,655 ERROR: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE

Build on <pyopencl.Device 'NVIDIA GeForce GTX 950M' on 'NVIDIA CUDA' at 0x196c900>:

(options: -D WORKGROUP_SIZE=1024 -I /usr/local/lib/python3.8/dist-packages/pyopencl/cl)
(source saved as /tmp/tmpjincpmum.cl)
2022-12-17 06:55:58,657 ERROR: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE

Build on <pyopencl.Device 'NVIDIA GeForce GTX 950M' on 'NVIDIA CUDA' at 0x196c900>:

(options: -D WORKGROUP_SIZE=1024 -I /usr/local/lib/python3.8/dist-packages/pyopencl/cl)
(source saved as /tmp/tmpg5j43wlq.cl)
2022-12-17 06:55:58,763 INFO: Reading data for image DJI_0649.JPG (queue-size=1
2022-12-17 06:55:58,765 INFO: Extracting ROOT_SIFT_GPU features for image DJI_0650.JPG
2022-12-17 06:55:58,851 ERROR: error while compiling sift: clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE - clBuildProgram failed: BUILD_PROGRAM_FAILURE

Build on <pyopencl.Device 'NVIDIA GeForce GTX 950M' on 'NVIDIA CUDA' at 0x196c900>:

(options: -D WORKGROUP_SIZE=1024 -I /usr/local/lib/python3.8/dist-packages/pyopencl/cl)
(source saved as /tmp/tmp3hvjwhfa.cl)
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

**If I use GPU, it's fine, but when I use GPU, I will get this error. So I think the dataset is OK.**

I enable the GPU acceleration by the instruction:

```
curl https://get.docker.com | sh   && sudo systemctl --now enable docker
distribution=$(. /etc/os-release;echo $ID$VERSION_ID)       && curl -fsSL https://nvidia.github.io/libnvidia-container/gpgkey | sudo gpg --dearmor -o /usr/share/keyrings/nvidia-container-toolkit-keyring.gpg       && curl -s -L https://nvidia.github.io/libnvidia-container/$distribution/libnvidia-container.list |             sed 's#deb https://#deb [signed-by=/usr/share/keyrings/nvidia-container-toolkit-keyring.gpg] https://#g' |             sudo tee /etc/apt/sources.list.d/nvidia-container-toolkit.list
sudo apt-get update
sudo apt-get install -y nvidia-docker2
sudo systemctl restart docker
sudo docker run --rm --gpus all nvidia/cuda:11.6.2-base-ubuntu20.04 nvidia-smi
```

I can see the GPU in the container:

```
root@W65-67SC:~# sudo docker run --rm --gpus all nvidia/cuda:11.6.2-base-ubuntu20.04 nvidia-smi
Sat Dec 17 07:06:13 2022       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.60.11    Driver Version: 525.60.11    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:01:00.0 Off |                  N/A |
| N/A   38C    P8    N/A /  N/A |      4MiB /  2048MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
```

Full console log attathed here:

[console.txt](https://github.com/OpenDroneMap/WebODM/files/10251150/console.txt)

pierotofy, 2022-12-17:
Your version of ODM is out of date; can you try to update to the latest (opendronemap/nodeodm:gpu) and try again?

chn-lee-yumi, 2022-12-17:
@pierotofy 

Thanks, this is due to the registry mirror I used. I don't know why it provides the image which is from 13 months ago. I pulled it from the docker hub directly and now it is fixed.

艹，阿里云的镜像加速服务神坑，居然拉了13个月前的镜像，害我排查半天以为代码问题。

closed 2022-12-17 by chn-lee-yumi