WebODM/WebODM #1283: Convert textured models PNG/TIFF textures to JPEG

pierotofy, 2023-01-11:
WebODM should automatically convert textured model textures to JPEG. It makes display and download faster.

pierotofy, 2023-01-11:
Hi @pierotofy,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2023-01-12 12:50) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

Saijin-Naib, 2023-01-11:
Would AVIF be in scope? Should compress better and supports transparency

pierotofy, 2023-01-11:
Nop, just JPEG.

pierotofy, 2023-01-25:
Closing in favor of https://github.com/OpenDroneMap/ODM/pull/1587

closed 2023-01-25 by pierotofy