WebODM/WebODM #1284: Newly add --single-texture obtain not available for the already processed datasets

muratmaga, 2023-01-13:
I have updated our WebODM docker installation to test this new feature. I can choose this for new datasets that I have, but the option is not there if I chose an existing (already processed data) and edit its option to create single texture.

Not a big deal for me, just wanted to notify.
@smathermather

muratmaga, 2023-01-13:
How did you install WebODM (docker, installer, etc.)?

Docker
What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Ubuntu 22.04, Firefox.
What is the problem?

I have updated our WebODM docker installation to test this new feature. I can choose this for new datasets that I have, but the option is not there if I chose an existing (already processed data) and edit its option to create single texture.

How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app/)

[Type answer here]

pierotofy, 2023-01-25:
Can't reproduce.

[image]

Before pressing "Edit", make sure you've selected a node that supports the `single-texture` option. That's probably why it didn't show up.

closed 2023-01-25 by pierotofy