WebODM/WebODM #1289: [api] Update project id permissions' from api

kikislater, 2023-01-31:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox but using python `requests` module

### What is the problem?

After reading documentation here https://docs.webodm.org/#update-a-project , I supposed to update a project though api. 
My main concern is to share project between others. Adding a new user, I usually need to update each project to set permissions of existing projects to new user with such values : 
```
[{'username': 'user1', 
'owner': True, 
'permissions': 
            ['add', 'change', 'delete', 'view']}, 
{'username': 'newuser', 
'owner': False, 
'permissions': 
            ['view']}
]
```
So I tried with the following request according to documentation 

```
>>> requests.patch(settings.SERVER + '/api/projects/171/permissions/', headers={'Authorization': 'JWT {}'.format(token)}, data={'username': user, 'owner': False, 'permissions': ['view']}).json()
{'detail': 'Method "PATCH" not allowed.'}
>>> requests.post(settings.SERVER + '/api/projects/171/permissions/', headers={'Authorization': 'JWT {}'.format(token)}, data={'username': user, 'owner': False, 'permissions': ['view']}).json()
{'detail': 'Method "POST" not allowed.'}
```
Of course I could read object permissions with `get` method

Update a project through api only allow to modify name and its description. Any plan to implement project permissions modification ?

pierotofy, 2023-01-31:
You can already do it, but the docs haven't been updated. See https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/components/EditPermissionsPanel.jsx#L43 (and help us update the docs?)

kikislater, 2023-02-01:
Hum ok thanks I saw that, so nothing to do with patch, I have to use post method.

But permissions are a bit confusing : 
If I run this, I get permission for my user
```
>>> requests.get(settings.SERVER + '/api/projects/40/', headers={'Authorization': 'JWT {}'.format(token)}).json()
{'id': 40, 'tasks': ['ba6acecb-ce39-4dae-9931-15d55d2d602f'], 'created_at': '2022-10-14T07:19:58.438999Z', 'permissions': ['add', 'change', 'delete', 'view'], 'name': 'Demo', 'description': ''}
```

Now If I run get to have project property on permissions I get this:
```
>>> requests.get(settings.SERVER + '/api/projects/40/permissions/', headers={'Authorization': 'JWT {}'.format(token)}).json()
[{'username': 'user', 'owner': True, 'permissions': ['add', 'change', 'delete', 'view']}]
```

If I want to add another user to my project I guess with  `/api/projects/40/edit` I update `/api/projects/40/` and not `/api/projects/40/permissions`
E.g.:
```
>>> requests.post(settings.SERVER + '/api/projects/40/edit/', headers={'Authorization': 'JWT {}'.format(token)}, data={'username': user, 'owner': 'False', 'permissions': 'view'}).json()
{'error': 'Invalid permissions'}
```

kikislater, 2023-02-01:
Btw if I made it through web browser, request from console give me on `/api/projects/40/edit` url :
```
{"name":"projectname","description":"","permissions":[{"username":"user1","owner":true,"permissions":["add","change","delete","view"]},{"username":"user2","owner":"false","permissions":["view"]},{"username":"user3","owner":"false","permissions":["view"]},{"username":"user4","owner":"false","permissions":["view"]},{"username":"user5","owner":"false","permissions":["view"]},{"username":"user6","permissions":["view"]}]}
```

Tried to replicate with python requests by removing user6 but got `{'error': 'Invalid permissions'}`, I missed something I don't catch ;-)

pierotofy, 2023-02-01:
Yep, I'd copy the requests from the web browser.

pierotofy, 2023-02-12:
Closing, feel free to update the docs if you think it might be useful for others. :pray:

closed 2023-02-12 by pierotofy