WebODM/WebODM #1294: Ploygon enhancements for volume measurements.

syazoo, 2023-02-22:
### How did you install WebODM (docker, installer, etc.)?
Windows Installer

[Firefox 110 on Windows 10]

### What is the problem?

When creating a polygon for volume measurement many times I accidentally complete the polygon if my click time is not longer than WebODM considers. I have tried to find a setting to change speeds including changing double click speed within windows settings. I am not sure what to suggest about the double click speed to close the poly, but I think that if the poly could be editted after it has been closed, then we could correct the mistake. The edit option also provides us a way to experiment or simply correct a mistake. The current option of deleting the poly and starting over can slow work flow and just be frustrating which is not the best state for doing good work.

Thank you very much. Please let me know if there is anything I can do including a contribution to fund.

pierotofy, 2023-02-22:
Hi @syazoo,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2023-02-23 09:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

syazoo, 2023-02-22:
What have I missed? I did use the template.

syazoo, 2023-02-22:
How did you install WebODM (docker, installer, etc.)?
Windows Installer
What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
[Firefox 110 on Windows 10]
What is the problem?

### What is the problem?

When creating a polygon for volume measurement many times I accidentally complete the polygon if my click time is not longer than WebODM considers. I have tried to find a setting to change speeds including changing double click speed within windows settings. I am not sure what to suggest about the double click speed to close the poly, but I think that if the poly could be edited after it has been closed, then we could correct the mistake. The edit option also provides us a way to experiment or simply correct a mistake. The current option of deleting the poly and starting over can slow work flow and just be frustrating which is not the best state for doing good work.

Thank you very much. Please let me know if there is anything I can do including a contribution to fund.
How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app/)
Draw a detailed poly around something like a stockpile. You will find that the more detailed you are that you select more points. In those moments of focus on the task, you will inevitably double click too fast at some point and close the poly. If you could edit the poly, then this issue is solved and drastically offers additional benefit of fine tuning or any other possible need or interest to improve your work.
[Type answer here]

I am not sure if this response is any different, the message saying it was incomplete after I used the template was a surprise.

pierotofy, 2026-10-29:
I think I will look into improving this soon. Sorry I didn't provide updates in a while.

Btw, since you mentioned funding @syazoo there's now a development fund at https://webodm.org/fund which might be of interest as it provides resources to work on features like these.