WebODM/WebODM #1298: WebODM freezing in Chrome with long filenames

ivangayton, 2023-03-10:
### How did you install WebODM (docker, installer, etc.)?

Docker install on Ubuntu 22.04

### What's your browser and operating system? 

Chrome 110 on Linux (Ubuntu 22.04)

### What is the problem?

When I upload a large batch of photos (a few thousand) with long descriptive filenames (e.g. `MyArea_Mission_15B_Flight_01_00002`) it begins uploading, and then Chrome freezes completely, requiring me to kill it from a console. When I restart Chrome, I can go back into the WebODM portal, I find that the task exists, but only contains some of the files (the first few hundred or thousand, depending how long the filenames were). 

__Workaround:__ I can rename the files to be much shorter but still unique (e.g. `15B_01_00002`) and the whole batch uploads without any issues. Sadly, _if I forget to do the same to my geo.txt file,_ my life becomes substantially less enjoyable as my task proceeds to undertake matching of thousands of photos without georeferencing (incidentally, some kind of warning that none of the lines in the geo.txt match any of the filenames might have saved me some heartache!

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I expect you can trigger the problem by attempting to upload a dataset with lots of photos and long filenames (if of interest, I can give you access to an FTP site with directories full of photos that have crashed in this fashion; they're a little big for GDriving).

Saijin-Naib, 2023-03-10:
Thanks for the report!

Could you provide us with one such full filename and how many files were in that Task that crashed?

I'm interested in seeing what the total character count is.

Also, if possible, having the path or length of the path would be helpful as that is likely prepended to the filenames and would then count towards the total character count.

ivangayton, 2023-03-10:
Actually, that's a very interesting thought about the paths; I suppose an additional workaround would be to move the directory full of images into the home folder to shorten the full path.

Which would be a hell of a lot easier and less risky than batch renaming 6,000 files.

pierotofy, 2025-09-09:
I cannot reproduce this on my machine. I generated 5,000 images with filename length 20 and they uploaded. Maybe this is due to updates that have been done since 2023, maybe it's my machine that is different or has more memory, in any case I'm not really sure what I could do to potentially fix this. If Chrome struggles to do many uploads and is made worse by filename length, I doubt it's due to WebODM.

I will close this, it would be good to check if this issue is still happening (we haven't received other reports) and if so, it might be interesting to gather additional information such as whether the issue happens only on certain machines, whether it happens on different browsers on the same machine, and whether it's operating system dependent. 🙏

closed 2025-09-09 by pierotofy