WebODM/WebODM #1300: Fix docs build

pierotofy, 2023-03-14:
The docs build is currently broken.
```
Git error: command `git clone --bare --no-hardlinks --quiet --no-tags --depth 1
--single-branch --branch 4.x -- https://github.com/middleman/middleman.git
/opt/hostedtoolcache/Ruby/2.7.7/x64/lib/ruby/gems/2.7.0/cache/bundler/git/middleman-0aa2a412fad4ddc12db204bb84a86fee137bc28e`
has failed.
Revision 4.x does not exist in the repository
https://github.com/middleman/middleman.git. Maybe you misspelled it?
Error: Process completed with exit code [11](https://github.com/OpenDroneMap/WebODM/actions/runs/4418839955/jobs/7746487447#step:4:12).

chris-bateman, 2023-10-25:
@pierotofy Would this potentially explain why some doco is missing?
https://github.com/OpenDroneMap/WebODM/blob/6bc90025aada533b38af7181cebc29685e0c9af4/app/api/tasks.py#L172

Is there an expectation that I should be able to see this in the doco?
http://docs.webodm.org/#task

pierotofy, 2025-07-18:
Still broken. 

```
Run bundle exec middleman build
bundler: failed to load command: middleman (/opt/hostedtoolcache/Ruby/2.7.8/x64/bin/middleman)
/opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax/haml_monkey_patch.rb:4:in `<module:Haml>': Filters is not a module (TypeError)
/opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/haml-6.3.0/lib/haml/filters/base.rb:5: previous definition of Filters was here
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax/haml_monkey_patch.rb:3:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax/extension.rb:4:in `require'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax/extension.rb:4:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax.rb:5:in `require'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-syntax-3.2.0/lib/middleman-syntax.rb:5:in `block in <top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/extensions.rb:91:in `load'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/extensions.rb:127:in `block in load_settings'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/extensions.rb:125:in `each'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/extensions.rb:125:in `load_settings'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/extension_manager.rb:12:in `initialize'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/application.rb:263:in `new'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-core-4.6.1/lib/middleman-core/application.rb:263:in `initialize'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-cli-4.6.1/bin/middleman:49:in `new'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/middleman-cli-4.6.1/bin/middleman:49:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/bin/middleman:23:in `load'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/bin/middleman:23:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli/exec.rb:58:in `load'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli/exec.rb:58:in `kernel_load'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli/exec.rb:23:in `run'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli.rb:492:in `exec'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/vendor/thor/lib/thor/command.rb:28:in `run'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/vendor/thor/lib/thor/invocation.rb:127:in `invoke_command'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/vendor/thor/lib/thor.rb:527:in `dispatch'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli.rb:34:in `dispatch'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/vendor/thor/lib/thor/base.rb:584:in `start'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/cli.rb:28:in `start'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/exe/bundle:37:in `block in <top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/lib/bundler/friendly_errors.rb:117:in `with_friendly_errors'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/lib/ruby/gems/2.7.0/gems/bundler-2.4.22/exe/bundle:29:in `<top (required)>'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/bin/bundle:23:in `load'
	from /opt/hostedtoolcache/Ruby/2.7.8/x64/bin/bundle:23:in `<main>'
```