WebODM/WebODM #1301: WebODM polygon clipping tool not working

picameratk, 2023-03-15:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

Docker on Window 11

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Windows 11 and Microsoft Edge

### What is the problem?

When viewing in 3D I attempt to use the polygon 3D tool, by placing points on my map, and the inside and outside buttons do nothing.  Note that the volume clip works.  I ran a ./webodm.sh update and tried again as well. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

[Type answer here]

picameratk, 2023-04-09:
I have created a video of the problem.  https://youtu.be/kja8zFQFdJE

pierotofy, 2023-11-15:
Thanks for the report and video @picameratk :pray: this should be fixed with #1436

closed 2023-11-15 by pierotofy

picameratk, 2023-11-16:
I send you this via email because github was not sending me a verification code but...

The fix for Potree #1436 will be reflected in webodm or docker desktop?  I have update both to the latest levels and still have the problem so I may  have expected the fix to be embedded too soon.

PR #1436 (2023-11-15): Fix Potree polygon clipping
Fixes #1301