WebODM/WebODM #1302: For anonymous non-logged-in users: WebODM should not show the navigation bar, specifically GCP functionality

tariqislam, 2023-03-19:
### How did you install WebODM (docker, installer, etc.)?

I cloned the repo, and used ./webodm.sh script to start it.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Mac OSX Ventura 13, Google Chrome

### What is the problem?

Problem:
SEEN - Even without logging in, one can access the GCP interface:
https://mapper.sltaeronautics.com/plugins/posm-gcpi/
EXPECTED - Except for a login screen, nothing else should be shown to an anonymous user, e.g.:
https://mapper.sltaeronautics.com/

More expected behavior details:
I’m running a pull from WebODM master branch as of a week ago, 3.0.5 I believe. So far the [posm-gcpi](https://mapper.sltaeronautics.com/admin/app/plugin/posm-gcpi/change/) and [diagnostic](https://mapper.sltaeronautics.com/admin/app/plugin/diagnostic/change/) I tested can be seen without being logged in once one opens this link: https://mapper.sltaeronautics.com/plugins/posm-gcpi/ : the sidebar menu shows up – in my opinion unless someone is logged in, no sidebar menu should show up. I don’t think WebODM has any features that are purely for anonymous users, right? And even if there is functionality to view a map totally anonymously, the sidebar navigation menu does not have items appropriate for an anonymous user.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

1. Without being logged in, go to the following url:
http://yourwebodm:8000/plugins/posm-gcpi/

pierotofy, 2023-03-19:
I get a 404 page when I access https://mapper.sltaeronautics.com/plugins/posm-gcpi/

Can you provide reproduction steps?

tariqislam, 2023-03-20:
Please check out the link now. I had disabled the plugin, hence it wasn't
showing up for the anonymous user.

Thanks.

Tariq

pierotofy, 2023-03-20:
Thanks, I've added a login check to the GCPI plugin in https://github.com/OpenDroneMap/WebODM/pull/1303/files

It's up to plugin developers to decide whether a plugin requires a log-in. Note anonymous access does not imply a security issue, but I'd say it makes sense for usability reasons in this case.

tariqislam, 2023-03-20:
Thank you for the quick fix! Does WebODM undergo any regular penetration
testing?

Tariq M. Islam
--
http://www.linkedin.com/in/tislammac