WebODM/WebODM #1314: Possible Bug: Import from old datasets instantly fail

Wlanfr3ak, 2023-03-28:
### How did you install WebODM (docker, installer, etc.)?

Ubuntu 22.04.2 with Docker Installation

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 111 on Windows 10

### What is the problem?

I created a new project and want to import an all.zip Dataset from another machine and its instantly fail after choose the file.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

tested on the master system (WebODM 1.9.16) and get no error, on the fresh installed (WebODM 2.0.1) every try.

[image]

Wlanfr3ak, 2023-03-28:
Ive try with another dataset was work instantly. Maybe a filesize problem?
Working File is the ~ 140 MB and not working File ist 4.3 GB

[image]

Saijin-Naib, 2023-03-28:
Are you able to share the datasets for testing?

Saijin-Naib, 2023-03-28:
I can't reproduce it locally with an 11.2GB dataset.

You're sure the archive is intact, you have ample disk space, and that the underlying filesystem supports files larger than 4GB?

closed 2023-03-30 by Wlanfr3ak