WebODM/WebODM #1316: Possible bug when processing large files?

tariqislam, 2023-03-30:
### How did you install WebODM (docker, installer, etc.)?

Docker

I have 3 physical machines. One runs both Webodm (no node) and clusterodm. The other two machines run only nodeodm.

ClusterODM is running with the following as the last commit:
commit 68b8cc8c528cbee5b7aa6db3ea0928fb6465ec8e (HEAD -> master, origin/master, origin/HEAD)
Merge: cf61924 172c683
Author: Stephen Mather <1174901+smathermather@users.noreply.github.com>
Date:   Thu Jan 26 18:51:47 2023 -0500

    Merge pull request #104 from nghi01/tunneling-hpc

    Adding instructions on tunneling to check if ClusterODM on HPC working or not
    
WebODM (without node) is running the following:
slt@nas:~$ sudo docker images
[sudo] password for slt:
REPOSITORY                   TAG         IMAGE ID       CREATED         SIZE
opendronemap/nodeodm         latest      a688a22049a6   4 months ago    1.85GB
redis                        latest      3358aea34e8c   4 months ago    117MB
opendronemap/webodm_webapp   latest      f681904052d1   5 months ago    1.65GB
opendronemap/clusterodm      latest      cfdb4ebea304   8 months ago    1.34GB
opendronemap/webodm_db       latest      e40c0f274bba   19 months ago   695MB
nvidia/cuda                  10.0-base   97cca2bac989   21 months ago   109MB

Processing Nodes are running (opendronemap/nodeodm   latest      9ffb1a462edf   7 days ago      1.94GB)

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Ubuntu 20
AMD Ryzen 7, 128GB RAM, 468GB swap, 2TB HDD

### What is the problem?

**May also be related to this issue in the community: https://community.opendronemap.org/t/any-guidance-for-large-area-ortho-maps/15303**

Seems to be a problem in processing large data sets. Only fast ortho seems to be working, and then again, not consistently. I have run a few other configurations changing the feature algo, triangulation, but all are giving very weird outputs if they complete. Jobs that I have stopped seem to be taking up all my ram (128GB) and the swap space growth keeps increasing until I finally terminate the job.

I have attached console outputs for a working fast ortho, a non-working default, and a non-working fast ortho.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

The failing data set is about 1000 photos, 1
[FastOrthoAndDSM-fail.pdf](https://github.com/OpenDroneMap/WebODM/files/11113312/FastOrthoAndDSM-fail.pdf)
[DefaultConsole-fail.pdf](https://github.com/OpenDroneMap/WebODM/files/11113313/DefaultConsole-fail.pdf)
[FastOrthoConsole-pass.pdf](https://github.com/OpenDroneMap/WebODM/files/11113314/FastOrthoConsole-pass.pdf)
0GB. 
I have previously run 4000 photos (40GB) on my system with default output configuration successfully -- took 14 hours to process.

pierotofy, 2023-03-30:
> If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app/)

closed 2023-03-30 by pierotofy

tariqislam, 2023-03-31:
@pierotofy Here is the link to the data:
https://drive.google.com/drive/folders/1gfxafR4jhB_G3vFAMA64b442MffBpJ-H?usp=sharing

Hopefully you can reopen the task?

I made another run with the following configuration:
auto-boundary: true, dsm: true, feature-quality: ultra, matcher-neighbors: 5, min-num-features: 20000, skip-3dmodel: true, rerun-from: dataset
It ran for 9 hours and completed successfully. 2D and 3D models are good, but the texture is non-existent -- only bits and pieces, and it doesn't align with the image.

pierotofy, 2023-03-31:
If it's a bug in WebODM, yes we can re-open it. Howver, often these errors are not due to bugs, but either issues with the images or with the settings.

Did you try process this dataset with the defaults? Did it work? A very low `matcher-neighbors` could be the problem.

tariqislam, 2023-03-31:
@pierotofy Yes, I have tried with defaults as well as fast ortho. I have already added details in the "What is the problem" and "How can we reproduce this" sections above, including attachments of console output for three runs. I can add more runs that failed as well, if needed.

pierotofy, 2023-03-31:
Default seems to have completed reading your log:

```
...
Compressing all.zip
Done!
```