WebODM/WebODM #1318: "Add Project" button should not be seen for a user who only has "view" permissions.

tariqislam, 2023-03-31:
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 111 on macOS (13.2)

### What is the problem?

Observed behavior:

This regards a logged in user who only has the following permissions:
app | Project | Can view Project
app | Task | Can view Task

When the user logs into WebODM, they can see the "Add Project" button. This button fails when clicked (project is not created), since the user does not have permissions to create a project, so the backend functionality is working as expected. 

Expected behavior:

The "Add Project" button should not be seen in the UI.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Details are above.

NtskwK, 2024-09-06:
Generally, the front-end gets the list of permissions from the api. It's a good way to use it to determine which components need to be rendered. But we don't have such an interface, do we need to create one?@pierotofy

pierotofy, 2024-09-06:
I personally would trickle the permission information from the view, simply because it avoids another request to the server, like we do for map view permissions.

https://github.com/OpenDroneMap/WebODM/blob/master/app/views/app.py#L78
https://github.com/OpenDroneMap/WebODM/blob/master/app/templates/app/map.html#L9-L11
https://github.com/OpenDroneMap/WebODM/blob/master/app/views/app.py#L50C10-L50C18

That requires some changes to the Dashboard instantiation for passing parameters from the view.

PR #1561 (2024-09-29): Hide Add Project button when user doesn't have permissions
Closes #1318
closed 2024-09-29 by pierotofy