WebODM/WebODM #1324: Cannot upload the following all assets zip file.

tariqislam, 2023-04-03:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Chrome 111 on macOS (13.3)](https://www.whatismybrowser.com/detect/what-version-of-chrome-do-i-have)

### What is the problem?

The following all_assets.zip (see Google Drive link below) was downloaded from one WebODM instance. It is fairly large, at about 4GB.

When I try to import it into a project in another WebODM instance running on my local laptop, it fails: 

[image]

It seems like the transfer of data worked, but the backend when it starts processing it that fails. Here is the log trace from the WebODM where it was being imported to:

`webapp             | WARNING Cannot get SETTINGS key from context. Something's wrong in settings_image_url.
webapp             | WARNING Cannot get SETTINGS key from context. Something's wrong in settings_image_url.
webapp             | WARNING Cannot get SETTINGS key from context. The footer will not be displayed.
webapp             | ERROR Internal Server Error: /api/projects/1/tasks/import
webapp             | Traceback (most recent call last):
webapp             |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/exception.py", line 34, in inner
webapp             |     response = get_response(request)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/base.py", line 115, in _get_response
webapp             |     response = self.process_exception_by_middleware(e, request)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/django/core/handlers/base.py", line 113, in _get_response
webapp             |     response = wrapped_callback(request, *callback_args, **callback_kwargs)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
webapp             |     return view_func(*args, **kwargs)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/django/views/generic/base.py", line 71, in view
webapp             |     return self.dispatch(request, *args, **kwargs)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 509, in dispatch
webapp             |     response = self.handle_exception(exc)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 469, in handle_exception
webapp             |     self.raise_uncaught_exception(exc)
webapp             |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 480, in raise_uncaught_exception
webapp             |     raise exc
webapp             |   File "/usr/local/lib/python3.9/dist-packages/rest_framework/views.py", line 506, in dispatch
webapp             |     response = handler(request, *args, **kwargs)
webapp             |   File "/webodm/app/api/tasks.py", line 441, in post
webapp             |     copyfileobj(file, fd)
webapp             | OSError: [Errno 0] Error
`

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Try to import the file linked here to a WebODM instance:
https://drive.google.com/drive/folders/1UqrkZ4VZH0B9ZrRAosDWA28HjLiK1EZo?usp=share_link

Wlanfr3ak, 2023-04-05:
I had an error like this, dont rename the File! Ive tested at with download again and without change to upload it and it works then see here: https://github.com/OpenDroneMap/WebODM/issues/1314

tariqislam, 2023-04-13:
Thanks for the comment. I'll close this issue as I have bee nplaying around with imports but I can't reproduce this anymore.