WebODM/WebODM #1326: WARNING Session data corrupted

ichsan2895, 2023-04-15:
This is happen after I updated to newest WebODM

```
# download newest git then extracted it to a folder named "WebODM-master"
$ cd WebODM-master

# update WebODM with GPU supports
$ ./webodm.sh update --gpu
01:00.0 3D controller: NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
GPU_NVIDIA has been found
Checking for docker...   OK
Checking for docker-compose...  OK
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.gpu.nvidia.yml -f docker-compose.nodemicmac.yml down --remove-orphans
Removing webapp                   ... done
Removing worker                   ... done
Removing webodm-master_node-odm_1 ... done
Removing db                       ... done
Removing broker                   ... done
Removing network webodm-master_default
Updating WebODM...
Skipping source update (.git directory not found)
docker-compose -f docker-compose.yml -f docker-compose.nodeodm.gpu.nvidia.yml pull
Pulling broker   ... done
Pulling db       ... done
Pulling worker   ... done
Pulling node-odm ... done
Pulling webapp   ... done
Done! You can now start WebODM by running ./webodm.sh start

# start it with GPU
$ ./webodm.sh start --gpu
01:00.0 3D controller: NVIDIA Corporation GP108M [GeForce MX150] (rev a1)
GPU_NVIDIA has been found
Checking for docker...   OK
Checking for docker-compose...  OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
Postgres DB directory: dbdata
SSL: NO
SSL key: 
SSL certificate: 
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.gpu.nvidia.yml up --scale node-odm=1
Creating broker                   ... done
Creating db                       ... done
Creating webodm-master_node-odm_1 ... done
Creating worker                   ... done
Creating webapp                   ... done
Attaching to broker, db, webodm-master_node-odm_1, worker, webapp
broker      | 1:C 15 Apr 2023 00:37:12.930 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 15 Apr 2023 00:37:12.930 # Redis version=7.0.10, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 15 Apr 2023 00:37:12.930 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
db          | 
db          | PostgreSQL Database directory appears to contain a database; Skipping initialization
db          | 
db          | LOG:  database system was shut down at 2023-04-15 00:36:31 UTC
broker      | 1:M 15 Apr 2023 00:37:12.930 * monotonic clock: POSIX clock_gettime
broker      | 1:M 15 Apr 2023 00:37:12.931 * Running mode=standalone, port=6379.
db          | LOG:  MultiXact member wraparound protections are now enabled
node-odm_1  | (node:1) NOTE: We are formalizing our plans to enter AWS SDK for JavaScript (v2) into maintenance mode in 2023.
[... 114 lines cut ...]
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMNMMMMMMMMM   ichsan@ichsan 
MMMMMMMMMMNs..yMMMMMMMMMMMMMm: +NMMMMMMM   ------------- 
MMMMMMMMMN+    :mMMMMMMMMMNo` -dMMMMMMMM   OS: MX x86_64 
MMMMMMMMMMMs.   `oNMMMMMMh- `sNMMMMMMMMM   Host: Aspire E5-476G V1.05 
MMMMMMMMMMMMN/    -hMMMN+  :dMMMMMMMMMMM   Kernel: 5.10.0-5mx-amd64 
MMMMMMMMMMMMMMh-    +ms. .sMMMMMMMMMMMMM   Uptime: 11 hours, 9 mins 
MMMMMMMMMMMMMMMN+`   `  +NMMMMMMMMMMMMMM   Shell: bash 5.0.3 
MMMMMMMMMMMMMMNMMd:    .dMMMMMMMMMMMMMMM   Resolution: 1366x768 
MMMMMMMMMMMMm/-hMd-     `sNMMMMMMMMMMMMM   DE: Plasma 5.14.5 
MMMMMMMMMMNo`   -` :h/    -dMMMMMMMMMMMM   WM: KWin 
MMMMMMMMMd:       /NMMh-   `+NMMMMMMMMMM   Theme: Breeze [Plasma], Breeze [GTK2/3] 
MMMMMMMNo`         :mMMN+`   `-hMMMMMMMM   Icons: Papirus [Plasma], breeze [GTK2/3] 
MMMMMMh.            `oNMMd:    `/mMMMMMM   Terminal: konsole 
MMMMm/                -hMd-      `sNMMMM   CPU: Intel i5-8250U (8) @ 3.400GHz 
MMNs`                   -          :dMMM   GPU: NVIDIA GeForce MX150 
Mm:                                 `oMM   GPU: Intel UHD Graphics 620 
MMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMM   Memory: 11568MiB / 23930MiB

$ docker images
REPOSITORY                        TAG         IMAGE ID       CREATED         SIZE
redis                             latest      33e3db53b328   2 days ago      117MB
opendronemap/nodeodm              gpu         9f35cf88538d   8 days ago      6.07GB
opendronemap/webodm_webapp        latest      96a8bb742eee   11 days ago     1.64GB
opendronemap/webodm_db            latest      e40c0f274bba   19 months ago   695MB
```

pierotofy, 2023-04-15:
Hi @ichsan2895,

It looks like there is some information missing from your issue that will be needed in order to diagnose and fix the problem at hand. Please take a look at the [Issue Template](https://github.com/OpenDroneMap/WebODM/blob/master/ISSUE_TEMPLATE.md), which will tell you **exactly** what your issue has to contain in order to be processable.

Also, double check that this is the right place. If you are just asking for information, reporting feedback or proposing a few feature, the right place to ask is the [Community Forum](http://community.opendronemap.org/c/webodm), not here.

I'm marking this one now as needing some more information. Please understand that if you do not provide that information within the next week (until 2023-04-15 20:55) I'll close this issue so it doesn't clutter the bug tracker.

Cheers!
~ Your friendly GitIssueBot

PS: I'm just an automated script, not a human being.

ichsan2895, 2023-04-15:
Sorry if I forget to mention something.

Regarding closed thread in https://github.com/OpenDroneMap/NodeODM/issues/202, I want to clarify:

[image]

Where is the menu as stated by error message?
[image]
There is no available `Administration > Processing nodes` menu

Also checked `Processing Nodes > node-odm-1`, but I don't find `Object Permissions` menu

pierotofy, 2023-04-15:
Could we move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! :+1:

ichsan2895, 2023-04-15:
> Could we move this conversation over to the forum at https://community.opendronemap.org? pray The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only). Thank you! +1

Great :+1: , I create this thread in there
https://community.opendronemap.org/t/warning-session-data-corrupted/15573

closed 2023-04-16 by pierotofy

ichsan2895, 2023-04-16:
Good news
Problem Solved

Here is the way to fix it

```
# remove all container that ODM has.
>> docker container rm -f  28ff90da39ae 9d83804a33a2 bfaecc7a272a 027e3cdaf0ae 9fcfa257f65d

# redownload WebODM with github clone
>> git clone https://github.com/OpenDroneMap/WebODM`

# install it again
>> cd WebODM
>> ./webodm.sh start --gpu
```