WebODM/WebODM #1328: Add Groups to projects like you can add users to projects.

dorminyk, 2023-04-17:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

Docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[Safari 16.4 on macOS (Catalina)](https://www.whatismybrowser.com/detect/what-version-of-safari-do-i-have)

### What is the problem?

No specific problem; Enhancement request. I'd like to be able to assign a project to a Group Easily from the dashboard.. Currently, you need to go into Administration, then Application, then projects, then Open your project and select "Object Permissions" in the upper right hand corner and then you have to know the name of the specific group. What I'd like to see the ability to add the "Group" to the Project just like you can assign someone to a project. It would make managing customer projects easier.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

N/A

dorminyk, 2023-04-17:
Added to enchantment log
https://community.opendronemap.org/t/add-groups-to-projects-like-you-can-add-users/15595

gotjoshua, 2024-06-14:
I fully agree.

I also think it would be very convenient to set a default group that all projects are part of.