WebODM/WebODM #1329: Nginx (1.18.0) that has known vulnerabilities. Would it be possible to upgrade Nginx or install patch for version 1.18.0?

aonwittawat, 2023-04-20:
### How did you install WebODM (docker, installer, etc.)?

The WebODM has been installed via docker file using suggested below command.
git clone https://github.com/OpenDroneMap/WebODM --config core.autocrlf=input --depth 1
cd WebODM
./webodm.sh start 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Edge 112 on Windows 11 [https://whatismybrowser.com/w/QDQQ3EM](https://whatismybrowser.com/w/QDQQ3EM)

### What is the problem?

The web server is using a version of Nginx (1.18.0) that has known vulnerabilities. See more details: [https://www.cybersecurity-help.cz/vdb/nginx/nginx/1.18.0/](https://www.cybersecurity-help.cz/vdb/nginx/nginx/1.18.0/). 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Upgrade Nginx to version 1.23.2+ or install patch from here: [http://nginx.org/en/security_advisories.html](http://nginx.org/en/security_advisories.html)

pierotofy, 2023-04-20:
Hey @aonwittawat , would you or your organization be willing to fund this?

pierotofy, 2023-04-27:
Closing as there doesn't seem to be interest to help with this.

closed 2023-04-27 by pierotofy