WebODM/WebODM #1340: Bilinear resampling for reprojecting DTM/DSM to custom EPSG

alpinfra, 2023-05-08:
### How did you install WebODM (docker, installer, etc.)?

Native Windows installer. Running Windows 10 Pro, Version 22H2

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

`https://whatismybrowser.com/w/TG2FBWY`

### What is the problem?

Vertical and horizontal lines (artefacts) in dtm/dsm data when downloading it in custom EPSG. Quite obvious when using a hillshade renderer for the dataset. I have stumbled upon this issue a couple of times outside of WebODM. It is caused by using Nearest Neighbour resampling technique when reprojecting continuous data such as terrain models. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Selecting custom EPSG, probably a none-UTM projection, for dtm/dsm data. See screenshot below:
[image]

**Recommended Solution:**

I suggest to use bilinear or cubic resampling for reprojecting Terrain or Surface data when selecting a custom EPSG. It does take more time than nearest neighbour, but eliminates the artefacts. As an example, I downloaded the data from the screenshot above in native UTM projection and reprojected the dataset with bilinear resampling to my selected EPSG. Cubic resampling also works. See screenshot below:
[image]

Further reading after a quick web search, maybe not the best literature: [https://gisgeography.com/raster-resampling/](url)

smathermather, 2023-05-17:
@pierotofy -- if you point me in the direction of the relevant code, I can take a look. I've got opinions on resampling especially of elevation... .

pierotofy, 2023-05-17:
https://github.com/OpenDroneMap/WebODM/blob/master/app/raster_utils.py#L167

Resampling probably needs to be changed when invoked for DEMs (only), although careful, `write_band` is used for orthos, plant health, colored DEMs, etc. so test, test, test if you change the logic.

evadregnum, 2023-05-18:
The same problem also exists with orthos. When exporting in UTM everything is ok, but when I use EPSG 2949, there are always squares in the orthos.
EPSG 2949
[image]
UTM
[image]

evadregnum, 2023-05-18:
ArcGis Pro 3 and ODM viewer. Directly in viewer of windows I see the lines

smathermather, 2023-05-18:
Yes, NN will leave artifacts in all continuous data.

To your point Piero: do you have reason to isolate any patch to resampling  to elevation models? Aerial imagery is continuous and would benefit from a more appropriate approach as well.

It is possible that different approaches might be more appropriate for elevation models than orthophotos (likely even), but I don't know this _a priori_.

Saijin-Naib, 2023-05-18:
Lanczos should work well for both

smathermather, 2023-05-19:
I have the concern that the resampling issues are distributed through the pipeline. I prefer Lanczos for most things most of the time: doesn't distort the location of features, compensates for the normal blurring associated with resampling with nominal sharpening, and generally does minimal distortion to statistics at most relevant scales.

The problem is that Lanczos also will emphasize artifacts that are extant from early portions of the pipeline. Every time I've seen a complaint about Lanczos, the culprit was actually the addition of it to the end of a pipeline without the removal of artifacts earlier in the pipeline.

So, the way I see it, we have a couple options: fix resampling everywhere it happens in WebODM and ODM, or use something like bilinear or cubic (or q1 or q3 -- never tried!) that will not emphasize and might slightly mask artifacts from earlier stages.

pierotofy, 2025-07-17:
This should be fixed with #1706. Update, try again and let me know if the issue persists? 🙏

closed 2025-07-17 by pierotofy