WebODM/WebODM #1341: addon: show CPU usage + Email notifications

pete1019, 2023-05-11:
Hi, would this be of interest? Should i provide a PR where the CPU Usage is also in diagnostic?

Had this issue where the CPU was at 3% but the project was still on working but it actually wasn't.
Think it was because there were pictures of a different project accidentally.

I thought of extending this with a plugin for email notification:

- on finish
- on error
- on low threshold CPU usage for quite some minutes.

Also, a checkmark in diagnostic to auto-refresh every X seconds.

What do you think?

pierotofy, 2023-05-12:
It might be useful to have a CPU indicator, yes! But might be tricky to get an accurate value that is cross-platform without additional dependencies.

An e-mail plugin would be most interesting!

Kathenae, 2023-05-16:
@pierotofy I'm working on the email notification plugin right now, I've managed to find the plugin signal for when a task has finished processing, but there doesn't seem to be one for when a task has failed? Any idea if such a signal exists if no, do you know the main points where a task may fail to process so i can send the plugin signal?

pierotofy, 2023-05-16:
Yep, there's currently no "task_failed" signal. It could be added though. See `app.plugins.signals.py` and `app.models.task`.

Kathenae, 2023-05-16:
Ohh alright, thanks!

pierotofy, 2023-05-19:
Done with #1343, #1344 :partying_face:

closed 2023-05-19 by pierotofy