WebODM/WebODM #1362: Exif tags reading error

arpitpara, 2023-07-15:
HI
I have already discussed this issue about my dataset on community platform.
I've not get any solution for the same. I am sure this problem is particular to WebODM and EXIF reading python package.
Can you give me a solution of this?
https://community.opendronemap.org/t/processing-error-with-rfly/16683

pierotofy, 2023-07-15:
Thanks for sharing a dataset @arpitpara 🙏 this should be fixed with https://github.com/OpenDroneMap/ODM/pull/1677

closed 2023-07-15 by pierotofy

arpitpara, 2023-07-15:
Should I update my webodm to get this fix?..
I bought windows license of it.

pierotofy, 2023-07-15:
You'll need to wait until an update for the Windows native version is released. If you're in a hurry, you can always make the modifications to `photo.py` manually by seeing what changed here https://github.com/OpenDroneMap/ODM/pull/1677/files

arpitpara, 2023-07-15:
Thanks for the update!
i know it's quite unprofessional way to ask to release a build early. But as a user I want to make request just consider it as ew need it quite soon.