WebODM/WebODM #1365: WARNING Memory overcommit must be enabled!

AnthonyCoughlin, 2023-07-25:
### How did you install WebODM (docker, installer, etc.)?

Docker (Ver 4.1.0)
### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome, Win 10 Pro

### What is the problem?

I'm receiving an error when starting Webodm:

_broker             | 1:M 25 Jul 2023 21:07:30.168 # WARNING Memory overcommit must be enabled! Without it, a background save or replication may fail under low memory condition. Being disabled, it can can also cause failures without low memory condition, see https://github.com/jemalloc/jemalloc/issues/1328. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect._

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Not sure how to reproduce the error.  I've restarted my machine...  I've updated WebODM.  I've tried the rebuild command and then updated again.  All with no success.  If I could get some guidance on the overcommit items mentioned above it would be greatly appreciated.  I've done some research, but I'm past my abilities.

Thanks in advance!

Saijin-Naib, 2023-07-26:
Did you edit the config file as indicated in the warning message?

After a reboot of the WSL2 backend, it should read that setting upon Docker startup and should proceed just fine.

AnthonyCoughlin, 2023-07-26:
I would love to edit the config file, but like I said above "I'm past my abilities"!  Would you mind telling me where I can find the config file?  I've been searching for "/etc/sysctl.conf" with no success!

Thanks again for your help!

Saijin-Naib, 2023-07-26:
Let's continue this on our Community, as the GitHub Issue tracker is for bug reports:  
https://community.opendronemap.org/invites/DGsLLWf1tr

closed 2023-07-26 by Saijin-Naib