WebODM/WebODM #1368: Contour line duplication

coreysnipes, 2023-08-15:
### How did you install WebODM (docker, installer, etc.)?

Windows installer.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 116 on Windows 10

### What is the problem?

Contour lines duplication when using Simplify > 0

### How can we reproduce this? 

- Process the Brighton Beach dataset using Default config or DSM+DTM config.
- Open Contours, using config of 1 meter contour, 0.2 simplify
- Zoom in to contours and observe that some contours have duplicate lines (i.e., two similar but distinct lines at a given altitude)

Observations:

- Does not seem to matter whether applying contour to DSM or DTM
- Most easily observed near the edges of the ortho, but not always limited to that region 
- Behavior *seems* to be limited to contours which do not extend all the way to the edge of the orthophoto region. i.e., drawing of contour turns back before reaching the edge of the ortho and continues drawing in the reverse direction
- Does not appear when Simplify = 0

Discussion and some photos here:  https://community.opendronemap.org/t/contour-duplicate-lines-when-using-simplify/17132/1

pierotofy, 2023-11-14:
@coreysnipes could you try again with the latest version? We recently changed the contour algorithm. 🙏

coreysnipes, 2023-11-15:
Yessir, will do!

closed 2023-11-15 by coreysnipes