WebODM/WebODM #1369: Error in Change Detection plugin (could not execute GRASS script changedetection.py)

coreysnipes, 2023-08-19:
### How did you install WebODM (docker, installer, etc.)?

Windows installer - v 2.0.3

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

firefox 116 on Win 10

### What is the problem?

Change detection plugin does not execute.  Dependency issue?  The error output is limited in the UI:

```
Could not execute GRASS script C:\WebODM\resources\app\apps\WebODM\coreplugins\changedetection\changedetection.py 
from C:\WebODM\resources\app\data\media\tmp\tmplddbrnxa_grass_engine:
```

When I try to blindly execute the script directly in powershell I show a missing module 'rasterio':

```
PS C:\WebODM\resources\app\apps\WebODM\coreplugins\changedetection> python .\changedetection.py
Traceback (most recent call last):
  File ".\changedetection.py", line 111, in <module>
    from compare import compare
  File "C:\WebODM\resources\app\apps\WebODM\coreplugins\changedetection\compare.py", line 1, in <module>
    import rasterio as rio
ModuleNotFoundError: No module named 'rasterio'
```

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

- Create two tasks with DSM+DTM
- Enable change detection plugin
- Attempt to use change detection on one task using default settings
- Observe error

Here are my change detection settings:

[image]

Here is the relevant section of app.log:

[change-detection-log.txt](https://github.com/OpenDroneMap/WebODM/files/12385610/change-detection-log.txt)

coreysnipes, 2023-08-19:
My python version is 3.7.9

Saijin-Naib, 2023-08-19:
Is this under the latest/current Build?

coreysnipes, 2023-08-19:
Hiya; here's the WebODM console output from powershell.  

Yes, I believe this is the latest build - 2.0.3 Build 89

[change-detection-webodm-log.txt](https://github.com/OpenDroneMap/WebODM/files/12385775/change-detection-webodm-log.txt)

Saijin-Naib, 2023-08-19:
Is this all of the log right from when you execute webodm.exe? It looks like a few lines are missing

pierotofy, 2023-08-20:
I'm guessing it's an issue with the installation of the necessary pip dependencies for the plugin on Windows.

pierotofy, 2023-12-18:
This is no longer applicable as the change detection plugin is now archived. https://github.com/OpenDroneMap/WebODM/pull/1435

closed 2023-12-18 by pierotofy