WebODM/WebODM #1370: Upload stuck on Pending

martinr36, 2023-08-19:
### How did you install WebODM (docker, installer, etc.)?

Windows Native

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Waterfox G6.0b2

### What is the problem?

The other week I uploaded an orthophoto to opendronemap using the plugin from within WebODM an d several hours later it was still showing as pending; so I posted about it on the WebODM forum https://community.opendronemap.org/t/image-uploaded-to-oam-still-processing/17107/6 and have now been asked to post here

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

All i did was upload it from my project using the plugin from within WebODM, the file concerned is in  the attached Zip, this is a screen shot I took at the time

[image]

And it is still sat doing the same 10 days later

[image]

I've just successfully uploaded another image from the same machine https://map.openaerialmap.org/#/-0.7290100000000077,51.623652013872395,18/user/634fce5e71072f000580f871/64e1526954fcac0001477573?_k=zmw9ff 

[odm_orthophoto.zip](https://github.com/OpenDroneMap/WebODM/files/12386865/odm_orthophoto.zip)

pierotofy, 2023-08-20:
The core issue is that the OAM plugin has no "retry" capabilities, and if for some reason the upload fails or gets stuck, it will remain stuck on pending.

kikislater, 2023-09-26:
And nothing is visible in logs. May be it could be interesting to log it !

kikislater, 2023-09-27:
After several tries, It doesn't seem ODM script could be culprit. OAM platform is not reliable even with webgui : a lot of 403 errors or uploading fails. It's a bit weird

kikislater, 2023-09-27:
Related issue: https://github.com/hotosm/OpenAerialMap/issues/158

I got some good results with ngrok as described here:
https://jakecoppinger.com/2023/03/generating-aerial-imagery-with-your-iphones-lidar-sensor/#aioseo-exporting-and-preparing-the-model

pierotofy, 2024-09-28:
This should now be fixed. If the issue persists, feel free to re-open? :pray:

closed 2024-09-28 by pierotofy