WebODM/WebODM #1373: Orthophoto_tiles cannot be loaded using leaflet.js.

tianmaoyu, 2023-08-23:
### How did you install WebODM (docker, installer, etc.)?

docker

opendronemap/nodeodm:3.1.9

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

chrome

### What is the problem?

I tried to load the tile images corresponding to the orthophoto generated by (WebODM nodeODM) using both leaflet.js and openlayers.js, but they failed to load. However, I was able to use tile images corresponding to orthophotos generated by other software. I suspect there may be a problem with nodeODM's generation of tile images for orthophotos.

#### 1 Generated tile images.

orthophoto_tiles： [orthophoto_tiles4.zip](https://github.com/OpenDroneMap/WebODM/files/12414855/orthophoto_tiles4.zip)

[image]

#### 2 tilemapresource.xml
```xml
<?xml version="1.0" encoding="utf-8"?>
    <TileMap version="1.0.0" tilemapservice="http://tms.osgeo.org/1.0.0">
      <Title>odm_orthophoto.tif</Title>
      <Abstract></Abstract>
      <SRS>EPSG:3857</SRS>
      <BoundingBox minx="114.18106547529626" miny="22.59195209633162" maxx="114.18370139110399" maxy="22.59502270654458"/>
      <Origin x="114.18106547529626" y="22.59195209633162"/>
      <TileFormat width="256" height="256" mime-type="image/png" extension="png"/>
      <TileSets profile="mercator">
        <TileSet href="5" units-per-pixel="4891.96980937500030" order="5"/>
        <TileSet href="6" units-per-pixel="2445.98490468750015" order="6"/>
        <TileSet href="7" units-per-pixel="1222.99245234375007" order="7"/>
        <TileSet href="8" units-per-pixel="611.49622617187504" order="8"/>
        <TileSet href="9" units-per-pixel="305.74811308593752" order="9"/>
        <TileSet href="10" units-per-pixel="152.87405654296876" order="10"/>
        <TileSet href="11" units-per-pixel="76.43702827148438" order="11"/>
        <TileSet href="12" units-per-pixel="38.21851413574219" order="12"/>
        <TileSet href="13" units-per-pixel="19.10925706787109" order="13"/>
        <TileSet href="14" units-per-pixel="9.55462853393555" order="14"/>
        <TileSet href="15" units-per-pixel="4.77731426696777" order="15"/>
        <TileSet href="16" units-per-pixel="2.38865713348389" order="16"/>
        <TileSet href="17" units-per-pixel="1.19432856674194" order="17"/>
        <TileSet href="18" units-per-pixel="0.59716428337097" order="18"/>
        <TileSet href="19" units-per-pixel="0.29858214168549" order="19"/>
        <TileSet href="20" units-per-pixel="0.14929107084274" order="20"/>
        <TileSet href="21" units-per-pixel="0.07464553542137" order="21"/>
      </TileSets>
    </TileMap>
    
```
#### 3  leaflet.js  code

```js
<!DOCTYPE html>
<html>
  <head>
    <meta charset="utf-8" />
    <title>test</title>
    <link
      rel="stylesheet"
      href="https://cdn.jsdelivr.net/npm/leaflet@1.9.4/dist/leaflet.min.css"
    />
    <style>
      #map {
        width: 100%;
        height: 100vh;
      }
    </style>
  </head>
  <body>
    <div id="map"></div>
    <script src="https://cdn.jsdelivr.net/npm/leaflet@1.9.4/dist/leaflet.min.js"></script>

    <script>
     
      var map = L.map("map", {
        crs: L.CRS.EPSG3857,
      }).setView([22.59195209633162, 114.18106547529626], 16);
      
      L.tileLayer("orthophoto_tiles4/{z}/{x}/{y}.png", {
        maxZoom: 21,
        minZoom: 5,
        tileSize: 256,
        // zoomOffset: -1,
      }).addTo(map);
    </script>
  </body>
</html>

```

####  Cannot load properly.
>  Tile images corresponding to orthophotos generated by other software can be used normally.

[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

1. Upload some drone images and set tiles to true;
2. Use leaflet.js to load the generated tile images.

pierotofy, 2023-08-23:
You need to set `tms: true` when creating a layer in leaflet. (This doesn't look like a bug, so I'm going to close it).

For future questions, let's move the conversation over to https://community.opendronemap.org? 🙏

closed 2023-08-23 by pierotofy

tianmaoyu, 2023-08-23:
I recently started learning about this technology and I am not familiar with a lot of it. Could you explain in more detail? How should I proceed? Where should I set tms: true?
@pierotofy

tianmaoyu, 2023-08-23:
thanks
@pierotofy 
```js
 <script>
     
      var map = L.map("map", {
        crs: L.CRS.EPSG3857,
      }).setView([22.59195209633162, 114.18106547529626], 16);
      
      L.tileLayer("orthophoto_tiles4/{z}/{x}/{y}.png", {
        maxZoom: 21,
        minZoom: 5,
        tileSize: 256,
        tms: true
        // zoomOffset: -1,
      }).addTo(map);
    </script>
```