WebODM/WebODM #1376: Error with broker on webodm.sh start / pinning docker images

vpicavet, 2023-09-04:
### How did you install WebODM (docker, installer, etc.)?

Linux docker installation throuhg `webodm.sh start`

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Linux Ubuntu Jammy / Lunar

### What is the problem?

WebODM installs correctly, but refuses to run, due to Redis not lauching, reporting a broken install ( most probably redis crashes ). Same behaviour on Ubuntu Jammy and on Lunar

### How can we reproduce this?

- Install fresh Ubuntu Jammy or Lunar and required dependencies for WebODM ( git, docker.io…)
- Follow WebODM installation instructions ( git clone & webodm.sh start )
- WebODM does not start

Error is exactly like the one reported here : 
https://community.opendronemap.org/t/broker-in-docker-cant-start-when-starting-over-git-bash/16416

----
```
 ./webodm.sh start
Checking for docker...   OK
Checking for docker-compose...  OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
Postgres DB directory: dbdata
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
Starting db                ... done
Starting webodm_node-odm_1 ... done
Starting broker            ... done
Starting worker            ... done
Starting webapp            ... done
Attaching to db, webodm_node-odm_1, broker, worker, webapp
broker      | 1:C 15 Jun 2023 16:52:22.738 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 15 Jun 2023 16:52:22.738 # Redis version=7.0.11, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 15 Jun 2023 16:52:22.738 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker      | 1:M 15 Jun 2023 16:52:22.738 * monotonic clock: POSIX clock_gettime
broker      | 1:M 15 Jun 2023 16:52:22.739 * Running mode=standalone, port=6379.
broker      | 1:M 15 Jun 2023 16:52:22.739 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
broker      | 1:M 15 Jun 2023 16:52:22.739 # Server initialized
broker      | 1:M 15 Jun 2023 16:52:22.739 # Fatal: Can't initialize Background Jobs.
db          |
db          | PostgreSQL Database directory appears to contain a database; Skipping initialization
db          |
db          | LOG:  database system was shut down at 2023-06-15 16:45:08 UTC
db          | LOG:  MultiXact member wraparound protections are now enabled
db          | LOG:  database system is ready to accept connections
db          | LOG:  autovacuum launcher started
node-odm_1  | (node:1) NOTE: We are formalizing our plans to enter AWS SDK for JavaScript (v2) into maintenance mode in 2023.
node-odm_1  |
node-odm_1  | Please migrate your code to use AWS SDK for JavaScript (v3).
node-odm_1  | For more information, check the migration guide at https://a.co/7PzMCcy
node-odm_1  | (Use `node --trace-warnings ...` to show where the warning was created)
[... 119 lines cut ...]
broker      | 1:M 15 Jun 2023 16:27:22.278 * Running mode=standalone, port=6379.
broker      | 1:M 15 Jun 2023 16:27:22.278 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
broker      | 1:M 15 Jun 2023 16:27:22.278 # Server initialized
broker      | 1:M 15 Jun 2023 16:27:22.278 # Fatal: Can't initialize Background Jobs.
broker      | 1:C 15 Jun 2023 16:27:22.946 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 15 Jun 2023 16:27:22.946 # Redis version=7.0.11, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 15 Jun 2023 16:27:22.946 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker      | 1:M 15 Jun 2023 16:27:22.947 * monotonic clock: POSIX clock_gettime
broker      | 1:M 15 Jun 2023 16:27:22.947 * Running mode=standalone, port=6379.
broker      | 1:M 15 Jun 2023 16:27:22.947 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
broker      | 1:M 15 Jun 2023 16:27:22.947 # Server initialized
broker      | 1:M 15 Jun 2023 16:27:22.947 # Fatal: Can't initialize Background Jobs.
broker      | 1:C 15 Jun 2023 16:27:24.023 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 15 Jun 2023 16:27:24.023 # Redis version=7.0.11, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 15 Jun 2023 16:27:24.023 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker      | 1:M 15 Jun 2023 16:27:24.023 * monotonic clock: POSIX clock_gettime
broker      | 1:M 15 Jun 2023 16:27:24.024 * Running mode=standalone, port=6379.
broker      | 1:M 15 Jun 2023 16:27:24.024 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
broker      | 1:M 15 Jun 2023 16:27:24.024 # Server initialized
broker      | 1:M 15 Jun 2023 16:27:24.024 # Fatal: Can't initialize Background Jobs.
broker      | 1:C 15 Jun 2023 16:27:25.895 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 15 Jun 2023 16:27:25.895 # Redis version=7.0.11, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 15 Jun 2023 16:27:25.895 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf

```

vpicavet, 2023-09-04:
### Possible cause and solution

It looks like latest Redis docker images have been generated with a recent version of Docker, leading to incompatibilities with versions of docker shipped with Ubuntu Jammy and Lunar ( 20.10.08 for lunar ). See https://github.com/redis/redis/issues/12362#issuecomment-1617561165 for hints.

A workaround is to reinstall docker with latest APT packages from the official Docker repository : version 24 solves the issue and WebODM installation is ok and runs correctly.

This is still a problem though, as the WebODM installation method on any ubuntu box is currently broken by default.

A good practice would be to pin Docker images versions used in WebODM, so as not to be dependent on problems raised by new images, and update versions carefully whenever fully tested.

pierotofy, 2023-09-04:
Thanks @vpicavet see if this is fixed with https://github.com/OpenDroneMap/WebODM/commit/f1b358db44b7db2e87e90d6c9f0f2f7ef4ea6b92 ? :pray:

vpicavet, 2023-09-04:
It will probably fix future problems, but not this one I guess. Before upgrading docker, I tried to pin redis to latest 7 and 6 versions, without success. I am not sure they still have versions of containers available on the hub ,generated with docker < 20.10.10. Therefore, not sure pinning dependencies will matter this time.

We can : 
- A/ work with upstream to see if they can fix the problem
- B/ update WebODM documentation to mention docker version > 20.10.10 is required

B seems the easiest and most straightforward option.

pierotofy, 2023-09-04:
`webodm.sh` could be updated to check for the docker version.

pierotofy, 2024-05-31:
I'm going to close this, as a workaround (update docker) is already available.

closed 2024-05-31 by pierotofy