WebODM/WebODM #1382: GRVI Vegetation Index values revision

jkairo, 2023-09-08:
### How did you install WebODM (docker, installer, etc.)?

Windows by an installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Browser: Firefox 117  Windows 10 Pro 22H2. But the issue is seen in the webodm which runs in the windows app (WebODM 2.0.3.)

### What is the problem?

I am using the WebODM to obtain the vegetation indexes from a drone orthophoto of a maize crop. According to literature **(Applying high-resolution visible-channel aerial imaging of crop canopy to precision irrigation management / [Redirecting 1](https://doi.org/10.1016/j.agwat.2019.02.017))**  the range for the Green Red vegetation index (GRVI ) is from -1 to 1 being, green vegetation (index above 0), water and snow (index around 0), and soils (index below 0) .

But the reading I am doing to the index color image and its histogram shows me the greenest for those pixels for water and soil and the most red for those where the vegetation is abundant (besides the river -the dark green in the upper and right side of the image), letting me the yellowish for the maize crop with a gradient

- imagen
[imagen1004×673 82.9 KB](https://community.opendronemap.org/uploads/default/original/2X/5/5e18cc28285b9721ead1f4b98244dbe06655ac08.jpeg)

I might change (invert) the color palette and have a congruent relation to the eye between color and elements of the image (red for soil and green for vegetation). But then is another problem if I want to do the analysis regarding the literature fundamentals, the numbers don´t match as the range of the GRVI states.

Then, in the community forum, Saijin_Naij asked me, What are the bands of your sensor? Your filter shows RGN, so Red/Green/NDVI. Is that what you actually have? 

So I showed him:

The sensor is RGB, it is not multispectral, nor thermal. The filters are by default in the Webodm, as you can see in the attached image. Is this filter affecting the pixel output number?

Sop he suggests that the filters should be normalized. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

https://hub.dronedb.app/r/tyto-agro/cerrito01_20230321

I use the plant health tool and select GRVI - I download the raw image and work directly in QGIS.

TimeScience, 2023-09-10:
[1-s2.0-S037837741831758X-main.pdf](https://github.com/OpenDroneMap/WebODM/files/12568287/1-s2.0-S037837741831758X-main.pdf)
Paper attached for reference since it is paywalled

jkairo, 2023-10-23:
hi, any news on this topic? is there any possible solution while the fault is corrected in the next or future version? Thanks in advance.

pierotofy, 2025-03-30:
Sorry it took me a while to look into this; the filter does indeed change the output values. But newer versions of WebODM have an "Auto" filter property which should automatically detect the proper band order.

I would recommend to update, try again and if the issue persists to attach a copy of your dataset (the screenshots / links in the original issue have expired). 🙏

closed 2025-03-30 by pierotofy

jkairo, 2025-04-22:
Hi Piero,

Thanks a lot for your effort and your support. I will give a try and will
tell you about. Meanwhile I found a solution with python thanks to IA. I
don't know how to code, just read some lines... anyway with the help of IA
I made this, I am sharing with you in case it may help you.