WebODM/WebODM #1383: WebODM should support uploading of .mp4 and .srt files

drewtu2, 2023-09-09:
### How did you install WebODM (docker, installer, etc.)?
via docker

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
[Safari 15.6 on macOS]

### What is the problem?
WebODM currently does not support the uploading of `.mp4` and videos, even though this functionality is now supported by ODM per this [PR](https://github.com/OpenDroneMap/ODM/pull/1567). It seems like there's a hard filter on the file extension types that can be uploaded via WebODM to just image types (.png, .jpg, and .svg). (My background is not in webdev, but I came across those in [`webpack.config.js`](https://github.com/OpenDroneMap/WebODM/blob/fd05b3a71f6485726052da9f32d9720615bfc24d/webpack.config.js#L67C8-L70), not sure if this is controlling that filter or not...)

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Create a new project and attempt to upload a video file and/or corresponding srt file. While the files will be visible in the file explorer, they will be un-selectable for upload.

drewtu2, 2023-09-09:
This might dupe https://github.com/OpenDroneMap/WebODM/issues/1356

pierotofy, 2023-09-09:
Mm, strange, we do have videos and SRT files in the accepted mime type list: https://github.com/OpenDroneMap/WebODM/blob/master/app/static/app/js/components/ProjectListItem.jsx#L142

Also seems to work fine on my Mac/Safari when I try to select a .mp4 file?

[image]

If you find a solution, feel free to open a pull request!

drewtu2, 2023-09-13:
ah. It appeared when I stopped and called 1`./webodm update`. I had pulled but didn't call the update so I must have been running with an old version.

closed 2023-09-13 by drewtu2