WebODM/WebODM #1384: Use COPC for streaming point cloud data instead of EPT

pierotofy, 2023-09-10:
Long term, it would be handy to stream COPC files directly to Potree for visualization, rather than generating EPT. The first step would be to write a contribution to Potree for a COPC reader.

TimeScience, 2023-12-14:
@gweis, is this something TERN might also be interested in and could potentially help develop?

pierotofy, 2023-12-14:
This actually landed in Potree 2 days ago: https://github.com/potree/potree/pull/1381

TimeScience, 2023-12-15:
That's great news!

What version of potree is ODM running? (I can't seem to find a version number in webODM)

pierotofy, 2023-12-15:
I think it's 1.8.0.

pierotofy, 2025-11-11:
No bandwidth for this. There's a lot of work involved to add this and I'm unsure of the benefits, aside from having less files for new projects, I have concerns about performance.