WebODM/WebODM #1388: Dasboard empty (all project disappeared) and cannot create new projects

SteCe68, 2023-09-12:
### How did you install WebODM (docker, installer, etc.)?

Installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome on Windows 11 home (Microsoft Surface Studio), do not see any answer to your question at the mentioned web address

### What is the problem?

All my projects disappear from my dashboard and get this message if I try to create a new one and get this error message: Could not load projects list: error
I restarted the PC and the software several times but nothing changed.

[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

No idea how to reproduce this problem

pierotofy, 2023-09-12:
Mm, @Saijin-Naib have you seen this one before?

Saijin-Naib, 2023-09-12:
Not in the wild, no, but I can think of a few ways it might happen.

@SteCe68 ,what antivirus (and any other security and cleaning software) do you have?

SteCe68, 2023-09-12:
> Not in the wild, no, but I can think of a few ways it might happen.
> 
> @SteCe68 ,what antivirus (and any other security and cleaning software) do you have?

Hi 
I have no antivirus installed on this PC at the moment (I had Kaspersky before but uninstalled it some months ago) and have only "standard" Windows security software.
Wort to mention that this happened after I forced the laptop to restart as it was not charging the battery anymore.
Adobe Lightroom is not starting at all after the forced-restart. May be the problem is with Windows.

What about uninstalling WebODM and installing it again? Will I see the old projects again or they will be lost?
I downloaded all orthophoto, DTM and cloud points I need but would prefer to have the projects in WebODM for a while

THanks
Stefano

Saijin-Naib, 2023-09-12:
Yep, this is what I was thinking.

You'll need to do a few advanced filesystem and Windows repair operations.

Are you up to doing such?

Reinstall on a filesystem and OS in an unhealthy state is not worth the effort, IMO.

SteCe68, 2023-09-13:
Thanks again
No I am not up to this, but if you can give me some guidance I can try to find somebody who that help me.

If not I guess I will have to Install everything from scratch, including windows?