WebODM/WebODM #1400: After upgrade to WebODM 2.1.2 build 100, lightning processes never complete.

ShammyLevva, 2023-09-21:
### How did you install WebODM (docker, installer, etc.)?

Local windows install processes running a Lightning process though.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Chrome 117.0.5938.89 Windows Server 2022

### What is the problem?

Ran two jobs one processing 945 images one processing 802 images. Both jobs submitted using Lightning server with options set to auto-boundary: true, crop: 2, dem-resolution: 2, dsm: true, orthophoto-resolution: 1, pc-quality: high

Both jobs sent an email saying completed but neither job has finished processing with the bar stuck as if it failed to finish around 80% mark. The timer has continued to increase for last several hours. I tried stopping WebODM and restarting it that then showed the jobs at the same percentage completion but the timers reset to the time it had taken to send the email so it's as if a final process after the email send is hanging. I am looking for information on what to do so that I can get the results of the processing. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

The problem was triggered after updating WebODM from v2.0.3 build 89 to v2.1.2 build 100. Jobs prior to that completed and finished as expected.

[image]

last few lines of one of the process logs are:

[INFO]    running /usr/bin/python3 "/code/opendm/tiles/hsv_merge.py" "/var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/odm_dem/dsm.previewcolor.tif" "/var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/odm_dem/dsm.previewhillshade.tif" "/var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/odm_dem/dsm.previewcolored_hillshade.tif"
0...10...20...30...40...50...60...70...80...90...100 - done.
[INFO]    running gdal_translate -outsize 1400 0 -of png "/var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/odm_dem/dsm.previewcolored_hillshade.tif" "/var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/opensfm/stats/dsm.png" --config GDAL_CACHEMAX 45.2%
Input file size is 1400, 1743
0...10...20...30...40...50...60...70...80...90...100 - done.
[INFO]    Exporting report to /var/www/data/c004003a-12c7-4037-b37b-4f3ca64a8279/odm_report/report.pdf
2023-09-20 21:25:17,020 DEBUG: Final doc sections size summary:
2023-09-20 21:25:17,020 DEBUG: - header.size: 9.0B
2023-09-20 21:25:17,023 DEBUG: - pages.size: 4.1KiB
2023-09-20 21:25:17,023 DEBUG: - resources.fonts.size: 199.0B
2023-09-20 21:25:17,208 DEBUG: - resources.images.size: 8.3MiB
2023-09-20 21:25:17,208 DEBUG: - resources.dict.size: 227.0B
2023-09-20 21:25:17,208 DEBUG: - info.size: 55.0B
2023-09-20 21:25:17,208 DEBUG: - catalog.size: 104.0B
2023-09-20 21:25:17,208 DEBUG: - xref.size: 610.0B
2023-09-20 21:25:17,208 DEBUG: - trailer.size: 67.0B
[INFO]    Finished odm_report stage
[INFO]    Running odm_postprocess stage
[INFO]    Post Processing
[INFO]    Finished odm_postprocess stage
[INFO]    No more stages to run
[INFO]    MMMMMMMMMMMNNNMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMNNNMMMMMMMMMMM
[INFO]    MMMMMMdo:..---../sNMMMMMMMMMMMMMMMMMMMMMMMMMMNs/..---..:odMMMMMM
[INFO]    MMMMy-.odNMMMMMNy/`/mMMMMMMMMMMMMMMMMMMMMMMm/`/hNMMMMMNdo.-yMMMM
[INFO]    MMN/`sMMMMMMMMMNNMm/`yMMMMMMMMMMMMMMMMMMMMy`/mMNNMMMMMMMMNs`/MMM
[INFO]    MM/ hMMMMMMMMNs.+MMM/ dMMMMMMMMMMMMMMMMMMh +MMM+.sNMMMMMMMMh +MM
[INFO]    MN /MMMMMMNo/./mMMMMN :MMMMMMMMMMMMMMMMMM: NMMMMm/./oNMMMMMM: NM
[INFO]    Mm +MMMMMN+ `/MMMMMMM`-MMMMMMMMMMMMMMMMMM-`MMMMMMM:` oNMMMMM+ mM
[INFO]    MM..NMMNs./mNMMMMMMMy sMMMMMMMMMMMMMMMMMMo hMMMMMMMNm/.sNMMN`-MM
[INFO]    MMd`:mMNomMMMMMMMMMy`:MMMMMMMNmmmmNMMMMMMN:`hMMMMMMMMMdoNMm-`dMM
[INFO]    MMMm:.omMMMMMMMMNh/  sdmmho/.`..`-``-/sddh+  /hNMMMMMMMMdo.:mMMM
[INFO]    MMMMMd+--/osss+:-:/`  ```:- .ym+ hmo``:-`   `+:-:ossso/-:+dMMMMM
[INFO]    MMMMMMMNmhysosydmNMo   /ds`/NMM+ hMMd..dh.  sMNmdysosyhmNMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMMs .:-:``hmmN+ yNmds -:.:`-NMMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMN.-mNm- //:::. -:://: +mMd`-NMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM+ dMMN -MMNNN+ yNNNMN :MMMs sMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM`.mmmy /mmmmm/ smmmmm``mmmh :MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM``:::- ./////. -:::::` :::: -MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMM:`mNNd /NNNNN+ hNNNNN .NNNy +MMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMd`/MMM.`ys+//. -/+oso +MMN.`mMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMMMMMy /o:- `oyhd/ shys+ `-:s-`hMMMMMMMMMMMMMMMMMM
[INFO]    MMMMMMMMNmdhhhdmNMMM`  +d+ sMMM+ hMMN:`hh-  sMMNmdhhhdmNMMMMMMMM
[INFO]    MMMMMms:::/++//::+ho    .+- /dM+ hNh- +/`   -h+:://++/::/smMMMMM
[INFO]    MMMN+./hmMMMMMMNds-  ./oso:.``:. :-``.:os+-  -sdNMMMMMMmy:.oNMMM
[INFO]    MMm-.hMNhNMMMMMMMMNo`/MMMMMNdhyyyyhhdNMMMM+`oNMMMMMMMMNhNMh.-mMM
[INFO]    MM:`mMMN/-sNNMMMMMMMo yMMMMMMMMMMMMMMMMMMy sMMMMMMMNNs-/NMMm`:MM
[INFO]    Mm /MMMMMd/.-oMMMMMMN :MMMMMMMMMMMMMMMMMM-`MMMMMMMo-./dMMMMM/ NM
[INFO]    Mm /MMMMMMm:-`sNMMMMN :MMMMMMMMMMMMMMMMMM-`MMMMMNs`-/NMMMMMM/ NM
[INFO]    MM:`mMMMMMMMMd/-sMMMo yMMMMMMMMMMMMMMMMMMy sMMMs-/dMMMMMMMMd`:MM
[INFO]    MMm-.hMMMMMMMMMdhMNo`+MMMMMMMMMMMMMMMMMMMM+`oNMhdMMMMMMMMMh.-mMM
[INFO]    MMMNo./hmNMMMMMNms--yMMMMMMMMMMMMMMMMMMMMMMy--smNMMMMMNmy/.oNMMM
[INFO]    MMMMMms:-:/+++/:-+hMMMMMMMMMMMMMMMMMMMMMMMMMNh+-:/+++/:-:smMMMMM
[INFO]    MMMMMMMMNdhhyhdmMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMMmdhyhhmNMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMNNNNNMMMMMMNNNNNNMMMMMMMMNNMMMMMMMNNMMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMh/-...-+dMMMm......:+hMMMMs../MMMMMo..sMMMMMMMMMMMM
[INFO]    MMMMMMMMMMMM/  /yhy-  sMMm  -hhy/  :NMM+   oMMMy   /MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMy  /MMMMN`  NMm  /MMMMo  +MM: .` yMd``` :MMMMMMMMMMMM
[INFO]    MMMMMMMMMMM+  sMMMMM:  hMm  /MMMMd  -MM- /s `h.`d- -MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMs  +MMMMM.  mMm  /MMMMy  /MM. +M/   yM: `MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMN-  smNm/  +MMm  :NNdo` .mMM` oMM+/yMM/  MMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMNo-    `:yMMMm      `:sNMMM` sMMMMMMM+  NMMMMMMMMMMM
[INFO]    MMMMMMMMMMMMMMMNmmNMMMMMMMNmmmmNMMMMMMMNNMMMMMMMMMNNMMMMMMMMMMMM
[INFO]    ODM app finished - Wed Sep 20 21:25:17  2023
100 - done.

pierotofy, 2023-09-21:
Hey @ShammyLevva :wave: thanks for the report.

When this happens again, could you attach a copy of the `app.log` file _without_ closing/restarting the app? You can access this by going to Tools --> View Logs. Note that if you already closed the app, wait for the problem to happen again, attaching a log after app reboot won't help us because it will get overridden.

ShammyLevva, 2023-09-22:
[app.zip](https://github.com/OpenDroneMap/WebODM/files/12703870/app.zip)

Reran one job same result email after about 4 hours that job has completed but showing around 80% complete. Zipped log file attached as requested.

closed 2023-09-22 by pierotofy

ShammyLevva, 2023-09-22:
Cheers for the speedy fix. Is there a new windows patch to download to fix the issue? I specifically purchased the windows exe version as I was getting nowhere fast with the pre-requisites for building it manually.

pierotofy, 2023-09-22:
I will publish an update for Windows by the end of the day :+1:

ShammyLevva, 2023-09-22:
Extremely good service many thanks for that.

Saijin-Naib, 2023-09-22:
Thanks for helping us confirm and reproduce! Can't do it without your contribution 🤓

ShammyLevva, 2023-09-22:
Many thanks update downloaded installed and I can confirm it fixed the issue. Happy Customer.

PR #1404 (2023-09-22): Fix console write encoding
- Fix encoding on Windows

Should fix #1400