WebODM/WebODM #1401: Cannot export correct multispectral values from DJI mavic3m

mape999, 2023-09-22:
### How did you install WebODM (docker, installer, etc.)?

Installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Firefox 117 on Windows 10

### What is the problem?

I have an issue with exporting multispectral images with webODM. I load my multispectral images, and I get the result, wich I donwload as a RAW image. When I load it in Qgis, I get a 5 band multispectral image. But the bands maximum values are very low (1.72594e-05), so when I try to create an NDVI image from the bands, with raster calculator it has absolutely false values.

It is interesting, because in the map viwer, it can display the values correctly, 1 - -1 (image below), however if I load it in qgis, I get an RGB image with values 0-255, or I can get a RAW image with values like 1.72594e-05.

NDVI display in web ODM:
[image]

NDVI values in Qgis:

[image]

[image]

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Dataset: https://drive.google.com/drive/folders/1Ra89v43Mt1tUrLLbgwbYuXacrK4_vOR3?usp=drive_link

For the processing, the default Multispectral option was chosen

pierotofy, 2023-09-22:
WebODM's NDVI index is capped between -1,1, the actual values in the RAW GeoTIFF are the ones displayed by QGIS. I don't think this is a bug? You certainly need to use the RAW GeoTIFF and not the RGB GeoTIFF if you want to compute indices in QGIS.

mape999, 2023-09-28:
I don't know if it's a bug, but I think that the NDVI values are not quite correct, processed by web ODM.

YMGong, 2024-04-20:
Hi I have exactly the same experience with using the matched monochromatic images in QGIS. Though I could derive the same (within the range of -1 to 1) NDVI in QGIS as in ODM because all the bands are scaled 'wrongly' the same way. But I wonder if this scaling issue is solved and what the scaling factor is... Because if I try to calculate any other indices in combination with the values from the RGB band it is just completely wrong...

golfredom, 2025-03-16:
Hello, having same issue, did you figure it out?

best

Sickculture, 2025-04-23:
I've the same problem. Any news?