WebODM/WebODM #1403: Feature Request: Rotate app and pgsql logs

Saijin-Naib, 2023-09-22:
### How did you install WebODM (docker, installer, etc.)?
Any

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
N/A

### What is the problem?
Support logs can be lost upon app restart unwittingly. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
Run WebODM, observe log contents, restart and note they are overwritten.

### Proposed Solution
Keep N most recent log copies for both app and pgsql, and name logs by timestamp of log opening so they can be sorted and identified.

pierotofy, 2025-02-06:
There's a few issues (including this one) that relate to the Windows installer version; we can keep these open, but I'd be nice to know if there are plans to dedicate resources to get them done?

Other one: #1469

Saijin-Naib, 2025-02-06:
I think that is the plan, yes.

Moving is fine, as well.