WebODM/WebODM #1418: Negative values in bands after calibration image?

ambarja, 2023-10-18:
Hello, I have recently been using images from the **Phantom 4 multispectral** with webodm, however when I perform the ND calibration to surface reflectance values with the :

 - Camera
 - Camera + sun

I have **negative value in the calibration bands**, this is very strange because the reflectance value is 0 - 1, right?
For this process I have been using [WebODM Lightning](https://webodm.net/)

Finally here put my image set used:
- [Set of P4M imagenes](https://drive.google.com/drive/folders/1piCcMsSKvpqWkcXIQ8Xp9IQ5OcTh7AT1?usp=share_link)

Here an example of the green band
[image]

pierotofy, 2023-10-21:
Did you set `texturing-skip-global-seam-leveling` while processing?

ambarja, 2023-10-21:
Hi @pierotofy , Thanks very much , I solved my error with this parameter. 
I close this issue for now.
[image]

closed 2023-10-21 by ambarja