WebODM/WebODM #1421: Wrong area measurement

Wilbodakwerken, 2023-10-23:
### How did you install WebODM (docker, installer, etc.)?

business installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

[[Chrome 118 on Windows 11](https://www.whatismybrowser.com/detect/what-version-of-chrome-do-i-have)

### What is the problem?

area measurement is not correct

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

When vertrical erea measurement is taken the oudput is not correct. 
For example i have a maesurement of 6,80m wide with a hight of 3,70m. This should give 25,16m². the software give me only 0,4m².

second example, a sloped roof thats 10,64m wide and 8,01 lenght it should be 85,23m², software tells me 77,2m²

Wilbodakwerken, 2023-10-24:
calculate the m² with the measurement and angels? 
below the screenshots, measurements are correct, m² isnt. 

[image]
[image]

pierotofy, 2023-11-14:
Should be fixed with #1435, thanks for the report! 🙏 

Note that it might take a few weeks before an update is released for Windows, so this fix will not be available right away.

closed 2023-11-14 by pierotofy