WebODM/WebODM #1445: Stitching of RGB and Multispectral Drone Images and Sensor Calibration Features

IbrahimUWA, 2023-12-11:
Hello,

I am very thankful to the WebODM team by developing a best stitching tool. I am working on RGB and Multispectral drone imagery for agricultural field analysis. I have successfully stitched both RGB and Multispectral images separately. However, when I try to stitch both RBB and multispectral images together as I require the blue channel from RGB for analysis, **the WebODM provides me the final stitched image without the RBG channel.  Please guide me in this regard. If it is possible to add a feature in WebODM to stitch both RGB and Multispectral images with all channels. Another important aspect in our analysis is to use the camera calibration features for more accurate analysis. I am requesting you if it is possible to add this facility as well. **

Regards
Muhammad Ibrahim
Research Scientist,
Department of Primary Industry and Regional Development, Western Australia.

pierotofy, 2023-12-11:
Hey @IbrahimUWA :wave: could you share a small subset of your images? (That can be reconstructed? Something like 20-60 images).

IbrahimUWA, 2023-12-12:
Hello Piero Toffanin,
Thank you so much for your kind reply. Please find the attached Zip file containing both RGB and Multi-spectral images. Please let me
know if you require more information.

Thank you.

Kind regards
Muhammad Ibrahim[https://res.cdn.office.net/assets/mail/file-icon/png/zip_16x16.png]RGBandMultispectral.zip<https://uniwa-my.sharepoint.com/:u:/g/personal/22675654_student_uwa_edu_au/EXtTFa86jNBJstTYBnjraKkBVS8E2eJBtJmfxWCs9ei3Ww>

IbrahimUWA, 2023-12-12:
Hello Piero Toffanin,
Thank you so much for your kind reply. Please find the attached Zip file containing both RGB and Multi-spectral images. Please let me
know if you require more information.

Thank you.

Kind regards
Muhammad Ibrahim

IbrahimUWA, 2023-12-12:
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123457_0001_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EZEjciciGzVJq3CR9bMJYikBAM7zGpBds6Bi-vbA0fL9Yg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123457_0001_MS_G.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EbTkg0wu6wJItR8gNrVHckQBrOL5ENMds3AdZ0dwhikr1w>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123457_0001_MS_NIR.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/Ed4Uxy_WTX1FmvPAcVZ0vv4Bj804Yc5OMcKBVjSEYuH8zw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123457_0001_MS_R.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/Eb4ak2CBOp5LmyuwtGtU-4YBOhQJrnHQJN9b1O9speyd0w>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123457_0001_MS_RE.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EcbrxeSjPnxJsD4CietOnPkBbf0WGIXkqdwLot1eS7apNg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123502_0002_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EeYk93nM80NEiJ7gB5lccMoBrJ148Q0sTxWhYNlSDkC8Yw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123502_0002_MS_G.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EcYn894WIcVJt5fIJYLMcdsBZ4v8vovpOjYZbHxEDipBSQ>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123502_0002_MS_NIR.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EaBKdfezsnpKvUXErphMKHIBOdcvNXL6zM8MCwpb5_Y3tQ>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123502_0002_MS_R.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EUQrzOCrY2hNjSCMHXN9QfABMTuv63sIhCZVedxyf2sGpw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123502_0002_MS_RE.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EScaQsjO4wZDnKAHHToS4QsB5gMwzmnRYsi1TANxw4Muuw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123507_0003_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EdSxDPZ4OWlAgEHD2m6aQGwBjbJs96XvIFoD6WK57khnSg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123507_0003_MS_G.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EV0gJTIQoBlCq1GU2Q2CsRABXyHCvCYCxIAVzN7Z2xp30A>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123507_0003_MS_NIR.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/ETqlIOdCODJIhydM3FxoFC8B6CxSz3aBIjMCTAKLmOxX8w>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123507_0003_MS_R.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/ERb8zsYF_YBLnDaHcNzdvtoBhV-I806k6eeC_l8Uwz6duA>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123507_0003_MS_RE.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EWUL_9LPKfVLuKISYAGLHC8BvFlbcsTe0CoHqA2ogfMWKg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123513_0004_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EZfcqwC1Dl1JqHV8ic1ISKsB8-9X9LemxlLNc-Kssyq6pg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123513_0004_MS_G.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EdLFHlsbbX9CosbY36m4pD8BVqK360vRwHsELtolpb2kww>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123513_0004_MS_NIR.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/ER4sVNyXW7tKmpKfyVF1jtYBStUwKY34c3RoAZICDYucVg>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123513_0004_MS_R.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EVY9JchjUKRIuMj5JTi8UZUBd-PnGfFyPmZF8WAA2OhWvw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123513_0004_MS_RE.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EbDpkTI3d9lBu7-kuXSHo24BRRiJLWkhOrsU5TSebzbNbQ>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123518_0005_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EdvIpTNcdcBEi_TduF9fX-sBqw509-MYR52XrkTEMwiVAA>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123518_0005_MS_G.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EU_BRgm-LTZNo6r2v7Fgwk8BJfM-9ndCR2WRXjhz8qdbiQ>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123518_0005_MS_NIR.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EbF8RTfrbGxDicpiNFBqVrUBFyrdSo7g0BP6SzgmBFYNOw>
[... cut ...] 413J-EBne1r4FvqsTXrrm2KCZywPw>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123529_0007_MS_RE.TIF<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EckOT3BqoF9Do4uiDVPDm1cBc-38nxHHpczOFAsUwoY3ig>
[https://res-h3.public.cdn.office.net/assets/mail/file-icon/png/photo_16x16.png]DJI_20230814123534_0008_D.JPG<https://uniwa-my.sharepoint.com/:i:/g/personal/22675654_student_uwa_edu_au/EeMpW_WjzDNCg0C_Cc9_POkBx_Jy4zgffFV__Koh6bPBXA>

pierotofy, 2023-12-12:
We can't access these:

```
<Error>
<Code>BlobNotFound</Code>
<Message>
The specified blob does not exist. RequestId:ec6bab3a-f01e-003a-07ba-2cd0b9000000 Time:2023-12-12T05:18:40.9931218Z
</Message>
</Error>
```

Can you share somewhere like Google Drive or Dropbox?

IbrahimUWA, 2023-12-12:
Hello,
The link below contains both unstitched images and stitched images. We did stitching separately for RGB and Multispectral separately.
https://drive.google.com/drive/folders/1tqn6JUUuM0AdgCST1XvVQIhK9ApwDzMd?usp=sharing
Google Drive: Sign-in<https://drive.google.com/drive/folders/1tqn6JUUuM0AdgCST1XvVQIhK9ApwDzMd?usp=sharing>
Access Google Drive with a Google account (for personal use) or Google Workspace account (for business use).
drive.google.com
 Let me know if you require further information.

Regards
Muhammad Ibrahim

pierotofy, 2023-12-14:
The issue with the Mavic 3M is that the images captured from the RGB sensor are too different than those captured with the multispectral sensor. So ODM drops the RGB band https://github.com/OpenDroneMap/ODM/blob/master/opendm/types.py#L120 when you try to process them all at once.

It might be possible to develop novel methods to process this dataset all at once (perhaps using OpenSfM's camera rigs function), but it's not a trivial fix.

IbrahimUWA, 2023-12-19:
> The issue with the Mavic 3M is that the images captured from the RGB sensor are too different than those captured with the multispectral sensor. So ODM drops the RGB band [...]

Hello Piero Toffanin,
Thank you so much for your response. Could you tell me how to apply Radiometric Calibration? I have camera calibrations photos and I know the Radiometric Calibration option on WebODM. But I have no idea how to use these photos for this. Please guide me regarding this, it will be highly appreciated.

Thanks

Kind regards
Ibrahim

pierotofy, 2023-12-19:
Let's move this conversation over to the forum at https://community.opendronemap.org? :pray: The forum is the right place to ask questions (we try to keep the GitHub issue tracker for feature requests and bugs only).

pierotofy, 2024-09-28:
Support for M3E was improved some time ago. I'll close this for now.

closed 2024-09-28 by pierotofy