WebODM/WebODM #1450: bug

Rico195786, 2023-12-21:
### How did you install WebODM (docker, installer, etc.)?

installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Your version is: Edge 120 on Windows 11

### What is the problem?

it can't process data that i was able to process on Pix4D, DJI Terra, Metashape, and Maps Made Easy without an issue.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Run the data I guess.  It just says there's an issue with the data or the software can not deal with it.  It won't let me copy the [art that says developers need this info to fix bug.  Please advise.

pierotofy, 2023-12-21:
Can you share a link to Google Drive / Dropbox with the images / settings?

pierotofy, 2024-09-28:
No dataset, cannot reproduce. Closing.

closed 2024-09-28 by pierotofy