WebODM/WebODM #1452: node fails to start with TypeError: require(...) is not a function

kpoman, 2023-12-28:
### How did you install WebODM (docker, installer, etc.)?
docker images

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Google chrome, Windows 10 pro

### What is the problem?

The web app starts fine, however, the node fails with some nodejs error:

```
node-odm-1  | /var/www/config.js:21
node-odm-1  | let argv = require('minimist')(process.argv.slice(2));
node-odm-1  |                               ^
node-odm-1  |
node-odm-1  | TypeError: require(...) is not a function
node-odm-1  |     at Object.<anonymous> (/var/www/config.js:21:31)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
node-odm-1  |     at Module.load (internal/modules/cjs/loader.js:950:32)
node-odm-1  |     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
node-odm-1  |     at Module.require (internal/modules/cjs/loader.js:974:19)
node-odm-1  |     at require (internal/modules/cjs/helpers.js:101:18)
node-odm-1  |     at Object.<anonymous> (/var/www/index.js:21:16)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
node-odm-1 exited with code 1

```
So, node is offline and refuses to be started. 
### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

I downloaded from scratch and installed latest git client and docker apps

kpoman, 2023-12-28:
After deleting the image, redownloading it and restarting, it fails in the same way:

```
$ ./webodm.sh start
Checking for docker...   OK
Checking for docker-compose...  OK
Starting WebODM...

Using the following environment:
================================
Host: localhost
Port: 8000
Media directory: appmedia
Postgres DB directory: dbdata
SSL: NO
SSL key:
SSL certificate:
SSL insecure port redirect: 80
Celery Broker: redis://broker
Default Nodes: 1
Settings:
Worker memory limit:
Worker cpus limit:
================================
Make sure to issue a ./webodm.sh down if you decide to change the environment.

docker-compose -f docker-compose.yml -f docker-compose.nodeodm.yml up --scale node-odm=1
 node-odm Pulling
 6f172cdbcbef Already exists
 1ceb7add7f88 Already exists
 8554f052ebbe Already exists
 7271cd6b4eb9 Already exists
 b46905e67503 Already exists
 26b2d1fbca59 Already exists
 f34e6b3c6463 Already exists
 12c093045606 Already exists
 56965d0a0aab Already exists
 2c9f41f9e7f2 Already exists
 4f4fb700ef54 Already exists
 4530cba6f703 Already exists
 b9f8fc79b5c8 Already exists
 node-odm Pulled
 Container db  Created
 Container webodm-node-odm-1  Created
 Container broker  Created
 Container worker  Created
 Container webapp  Created
Attaching to broker, db, webapp, node-odm-1, worker
node-odm-1  | /var/www/config.js:21
node-odm-1  | let argv = require('minimist')(process.argv.slice(2));
node-odm-1  |                               ^
node-odm-1  |
node-odm-1  | TypeError: require(...) is not a function
node-odm-1  |     at Object.<anonymous> (/var/www/config.js:21:31)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
node-odm-1  |     at Module.load (internal/modules/cjs/loader.js:950:32)
node-odm-1  |     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
node-odm-1  |     at Module.require (internal/modules/cjs/loader.js:974:19)
node-odm-1  |     at require (internal/modules/cjs/helpers.js:101:18)
node-odm-1  |     at Object.<anonymous> (/var/www/index.js:21:16)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
broker      | 1:C 28 Dec 2023 21:50:59.358 # oO0OoO0OoO0Oo Redis is starting oO0OoO0OoO0Oo
broker      | 1:C 28 Dec 2023 21:50:59.358 # Redis version=7.0.10, bits=64, commit=00000000, modified=0, pid=1, just started
broker      | 1:C 28 Dec 2023 21:50:59.358 # Warning: no config file specified, using the default config. In order to specify a config file use redis-server /path/to/redis.conf
broker      | 1:M 28 Dec 2023 21:50:59.359 * monotonic clock: POSIX clock_gettime
broker      | 1:M 28 Dec 2023 21:50:59.359 * Running mode=standalone, port=6379.
broker      | 1:M 28 Dec 2023 21:50:59.359 # Server initialized
broker      | 1:M 28 Dec 2023 21:50:59.359 # WARNING Memory overcommit must be enabled! Without it, a background save or replication may fail under low memory condition. Being disabled, it can can also cause failures without low memory condition, see https://github.com/jemalloc/jemalloc/issues/1328. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
[... 74 lines cut ...]
node-odm-1  |                               ^
node-odm-1  |
node-odm-1  | TypeError: require(...) is not a function
node-odm-1  |     at Object.<anonymous> (/var/www/config.js:21:31)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
node-odm-1  |     at Module.load (internal/modules/cjs/loader.js:950:32)
node-odm-1  |     at Function.Module._load (internal/modules/cjs/loader.js:790:12)
node-odm-1  |     at Module.require (internal/modules/cjs/loader.js:974:19)
node-odm-1  |     at require (internal/modules/cjs/helpers.js:101:18)
node-odm-1  |     at Object.<anonymous> (/var/www/index.js:21:16)
node-odm-1  |     at Module._compile (internal/modules/cjs/loader.js:1085:14)
node-odm-1  |     at Object.Module._extensions..js (internal/modules/cjs/loader.js:1114:10)
node-odm-1 exited with code 1
Gracefully stopping... (press Ctrl+C again to force)
 Container webapp  Stopping
 Container webodm-node-odm-1  Killing
 Container worker  Killing
 Container webapp  Killing
 Container db  Killing
 Container broker  Killing
 Container webodm-node-odm-1  Killed
time="2023-12-28T18:51:04-03:00" level=error msg="got 3 SIGTERM/SIGINTs, forcing shutdown"

```

kpoman, 2023-12-29:
From what I see, there is a bug in this line:
`let argv = require('minimist')(process.argv.slice(2));
`
which should be:
```

var minimist = require('minimist');
let argv = minimist(process.argv.slice(2));
```

Could you please fix the image?
Thank you

pierotofy, 2023-12-29:
Moving the conversation over to https://github.com/OpenDroneMap/NodeODM/issues/211

closed 2023-12-29 by pierotofy