WebODM/WebODM #1453: Feature Request

duiveldoder, 2023-12-28:
The html gui doesn't automatically update.. so I keep pressing ctrl-F5 to update the RAM consumption but also other fields ..

Is there a possibility to include an button on the upper right or maybe in the settings, if checked, then the GUI updates automatically every 5 seconds or so?

MarkAtGit, 2024-10-23:
Maybe it's also nice to add the possibility to download some (basic) Diagnostic usage statistics from the last 12 hours, updated every minute (720 records) in this Diagnostic Plugin. The same way as you can now see the log file while processing projects.
[image]

Best regards,

Mark 
Aeret

pierotofy, 2025-10-16:
I'd welcome pull requests to improve the diagnostic plugin 🙏 if there's interest, feel free to open one?

closed 2025-10-16 by pierotofy

HikeAndMap, 2025-10-16:
I have to say, @jet86 I'll keep your development under observation.. I'm very happy to see what you're doing there