WebODM/WebODM #1456: "DroneDB plugin error".

EFPDroneFootage, 2023-12-30:
****************************************
PLEASE REMOVE THIS NOTE AFTER READING IT!

First of all, thank you for taking the time to report an issue.

Before you continue, make sure you are in the right place. Please open an issue only to report faults and bugs. For installation problems, questions, server configuration problems and feature requests, please open a topic on http://community.opendronemap.org/c/webodm. 

Also, is your problem listed here? https://github.com/OpenDroneMap/WebODM#common-troubleshooting
If so, do not open an issue. We already know about it and steps to solve it are already documented.

Every issue needs to follow the format specified below. Do not remove the questions, simply answer each question by typing below it. If you don't know how to answer a question, write so. Questions that don't follow these guidelines will be automatically closed by our friendly IssueBot (who's not very smart and needs to see a particular format to keep an issue open).
****************************************
### How did you install WebODM (docker, installer, etc.)?

Via windows install. Also WebODM Lightning. 

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

 Chrome 120 on Windows 11

### What is the problem?

Everytime I try to use the "Share to DroneDB" in WeboDM it fails with "DroneDB plugin error".

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

By having the DroneDB plugin activated, then being in the WebODM dashboard, selecting the task drop down, and where we see options such as Open Map, we see the Share to Drone DB button, this when pressed begins an action, sends a few KBs of data over to DroneDB, then says it has an error which reads - Error uploading file odm_report\shots.geojson: Failed to upload file.
[image]

EFPDroneFootage, 2023-12-30:
[image]

pierotofy, 2023-12-30:
cc @HeDo88TH

EFPDroneFootage, 2023-12-31:
Do I need to find WebODM Lighning desktop version 3 as I am on 2 now upgraded from 1.9 with the error and now I see the dev team has made a version 3 and ironed out the DroneDB issue but I can only find V2 in teh cloud site via WebODM.net somewhere but if you have a v3 where would I find it to install that and see if that works the DroneDB share button?

EFPDroneFootage, 2023-12-31:
> Do I need to find WebODM Lighning desktop version 3 as I am on 2 now upgraded from 1.9 with the error and now I see the dev team has made a version 3 and ironed out the DroneDB issue but I can only [...]

> cc @HeDo88TH

Do I need to find WebODM Lighning desktop version 3 as I am on 2 now upgraded from 1.9 with the error and now I see the dev team has made a version 3 and ironed out the DroneDB issue but I can only find V2 in teh cloud site via WebODM.net somewhere but if you have a v3 where would I find it to install that and see if that works the DroneDB share button?

pierotofy, 2023-12-31:
The WebODM Lightning Desktop version hasn't been updated in a while. A workaround in the meanwhile could be to use the cloud platform at https://webodm.net/cloud.

EFPDroneFootage, 2023-12-31:
When I buy the full install in January, will all my databases and links to
my client facing sharing server still be in place within it and if so I am
happy to wait to get to the updated version in Jan and then take it all
from there? Thank you for the advice and info. I love what you all do.

pierotofy, 2024-09-28:
This *might* have been fixed with the latest version.

If not, contact https://dronedb.app/contact for a time estimate on a fix.

closed 2024-09-28 by pierotofy