WebODM/WebODM #1461: DJI M3E Multispectral - Blue orthophoto

danieldobosi, 2024-01-19:
### How did you install WebODM (docker, installer, etc.)?

both standalone on linux and lightning.
linux version: WebODM 2.4.1

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

Edge 120 on Windows 10

### What is the problem?

When processing a full DJI M3E Multispectral dataset, the RGB orthophoto is blue

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

process these datasets to reproduce
https://e38surveysolutions.com/blogs/news/mavic-3-multispectral-data
https://www.pix4d.com/dji-mavic-3-multispectral-dataset/

parameters: “radiometric-calibration: camera” 

screenshots
[image]
[image]

pierotofy, 2024-01-19:
Seems the same issue as #1445

pierotofy, 2024-01-19:
Yes, that's because ODM will currently drop the JPG images and only keep G, R, NIR and Re. 

I'm working on a potential fix, I will share in the upcoming days..

pierotofy, 2024-01-19:
Ref: https://github.com/OpenDroneMap/ODM/pull/1740

smathermather, 2024-07-21:
Let us know if this isn't resolved with https://github.com/OpenDroneMap/ODM/pull/1771 and we can re-open. Thanks for reporting!

closed 2024-07-21 by smathermather

danieldobosi, 2024-07-31:
unfortuntately still blue:

[image]

I've also restarted two different datasets, same results

pierotofy, 2024-07-31:
The JPGs are still being dropped. See discussion on https://github.com/OpenDroneMap/ODM/pull/1771#issuecomment-2197639872