WebODM/WebODM #1463: WebODM point cloud interface not showing classifications of points in Potree

smathermather, 2024-01-23:
Original report here:
https://community.opendronemap.org/t/cant-see-point-cloud-classification-unless-viewing-exported-las-laz-file-in-potree/19005

### How did you install WebODM (docker, installer, etc.)?

I believe this to be independent of install type, but definitely happens with docker installs.

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

I am running on Ubuntu 20.04. Unsure the version from the reporter.

### What is the problem?

> I noticed that the point cloud 3D viewer does not distinguish the classified object in the point cloud. All points are “unclassified.”
> However if I export the point cloud to a LAS/LAZ file, and view that file with Potree, the object classification works properly.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Seems to happen on any dataset. DSM/DTM settings used on any geospatial dataset at https://github.com/OpenDroneMap/ODMData  should reproduce.

smathermather, 2024-01-23:
Silly github-actions bot. Who's a good bot‽

Re-opening.

joe-stroup, 2024-01-23:
I'm the person who identified this issue.

I have WebODM running both on my desktop, and on a Ubuntu server using Docker.

In both cases, when running a model where point cloud classification is performed (DSM and/or DTM created) the version of Potree that is embedded with WebODM does not have visibility to the point cloud classifications.  All the points in the point cloud are classified as "never classified."

However, if the point cloud is downloaded as a LAZ or LAS file, and that file is the loaded into an independent implementation of Potree, the point cloud classifications are correct.

Feel free to contact me for additional details if needed.

smathermather, 2024-01-23:
Awesome. Thanks for weighing in and identifying in the first place. Can you list your OS, install process, and what browser you're using for viewing?

joe-stroup, 2024-01-23:
My desktop implementation is Windows 10.  I installed both WebODM and Potree using their respective GitHub installation procedures in the readme.

My server implementation is a Ubuntu droplet (their current long term release) - the server has 16 intel cpu's and 64gb of storage.

I installed WebODM on this server using the GitHub readme for installing WebODM using Docker on a cloud service such as AWS, etc.

As I've experience this issue on both the desktop and server implementations, I think it should be easy to reproduce and verify on anyone else's recent installation.

joe-stroup, 2024-01-23:
Additionally, I'm experience the issue with both Chrome and Edge - whatever is their current versions.

joe-stroup, 2024-01-23:
Done - and I just added my browser info.

closed 2024-01-25 by pierotofy

Saijin-Naib, 2024-02-02:
https://github.com/OpenDroneMap/WebODM/issues/996

Closes mine, too 🤣