WebODM/WebODM #1469: Local Python Site-Packages Polluting PATH

Saijin-Naib, 2024-02-13:
### How did you install WebODM (docker, installer, etc.)?
Installer

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)
Windows

### What is the problem?
Some installs seem to never work on select machines (error 1 at start), unless a new userprofile is made. 

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)
User here seems to have isolated it to contention with packages in Site-Packages on the host. 
https://community.opendronemap.org/t/exit-code-1-cant-run-native-windows-software/19279/7?u=saijin_naib

This fits with seeing the issue on machines with dev and geospatial tooling installed.

Saijin-Naib, 2024-08-01:
Confirmed multiple times. 

Fix is removal of Python runtimes and removal of Python folder in user's %AppData%. 

Somehow, we are contending with packages there.

pierotofy, 2024-08-01:
Is there a way to reproduce it on a working machine?

Saijin-Naib, 2024-08-01:
Installing EXE/MSI python runtime, or Anaconda seems to do it. 

I do not believe MSIX/Store version interferes.

pierotofy, 2024-08-01:
If you can write exact steps (with machine specs info, which runtime version to install, which pip packages to install to break the software, etc.) I could try to take a look.

I ask because I've installed a different version of Python before and things were working fine at the time at least on my dev machine.