WebODM/WebODM #1470: 3d page screen on smartphone

kemaluzel, 2024-02-14:
When the webodm 3D model page is opened with the Chrome browser on the smartphone, the buttons such as share and 2D at the bottom right do not appear on the screen. The page cannot be scrolled down manually.

kemaluzel, 2024-02-14:
[image]
this is a minor bug

vdroners, 2024-02-17:
I also confirm this issue.

pierotofy, 2024-09-28:
Please try the latest version :pray: 

We've made significant improvements on the mobile UI/UX since this was first reported. :clinking_glasses: 

If the issue persists, feel free to re-open :pray:

closed 2024-09-28 by pierotofy

vdroners, 2024-09-29:
> Please try the latest version; we've made significant improvements on the
> mobile UI/UX since this was first reported. 🥂
>
> If the issue persists, feel free to re-open 🙏

Thank. You!

V/R

Daniel Sautter
Veteran Droners
(540) 848-7979