WebODM/WebODM #1471: Update Django Framework from 2.2.27 to 4.2 LTS

chris-bateman, 2024-02-16:
### How did you install WebODM (docker, installer, etc.)?

N/A

### What's your browser and operating system? (Copy/paste the output of https://www.whatismybrowser.com/)

N/A

### What is the problem?

Django framework is EOL.  WebODM uses 2.2.27 and the current LTS is 4.2
Placeholder issue to begin discussion on upgrading the framework.

### How can we reproduce this? (What steps trigger the problem? What parameters are you using for processing? Include screenshots. If you are having issues processing a dataset, you must include a copy of your dataset uploaded on Dropbox, Google Drive or https://dronedb.app)

Review Django version here - https://github.com/OpenDroneMap/WebODM/blob/8150b822d9851fca434a5dedbdb5c5e1b3da2c3d/requirements.txt#L7

4.2 LTS makes sense support-wise, but v5 is also out.

chris-bateman, 2024-02-16:
EOL notes - https://www.djangoproject.com/download/ and https://endoflife.date/django

Saijin-Naib, 2024-02-16:
Do you know of any gotchas? API breaks? 

Keeping current on supported dependencies would be wonderful.

chris-bateman, 2024-02-16:
So this doco covers the process - https://docs.djangoproject.com/en/5.0/howto/upgrade-version/

Example incompatible notes - https://docs.djangoproject.com/en/5.0/releases/4.2/#backwards-incompatible-changes-in-4-2

And a good list here as well - https://docs.djangoproject.com/en/5.0/internals/deprecation/

chris-bateman, 2024-03-03:
I started to make a little progress on this.  I am playing dependency wack a mole.
So with that in mind it might be time to audit the current modules located in this file - 
https://github.com/OpenDroneMap/WebODM/blob/master/requirements.txt

The main focus is identifying packages that are deprecated/abandoned with no clear support from their maintainer/community.
Some have suitable alternatives noted in their doco but others will require some research to find a suitable replacement.

**django-guardian**
[image]

**rfc3987** - Github repo was archived on the 12th Jan 2024 but moved to a new repo recently.
https://codeberg.org/atufi/rfc3987/
[image]

**djangorestframework-jwt** - Github archived 
Alternatives - https://github.com/jpadilla/django-rest-framework-jwt/issues/484
[image]

**itypes** - Github archived
[image]

**django-rest-framework-guardian** 
[image]

**strict-rfc3339**
[image]

**openapi-codec** - Github archived
[image]

**django-codemirror2** - After 7 years of no activity has had a few updates this year. 
[image]

**piexif**
[image]

smathermather, 2024-04-09:
This is great Chris. Anything you need to help, e.g. identifying alternatives? I would say, for anything that has a known upgrade path, feel free to upgrade and test, and send me a relevant repo to review and I'm happy to run in production on my end and help find any issues.

NtskwK, 2024-07-21:
> Do you know of any gotchas? API breaks?
> 
> Keeping current on supported dependencies would be wonderful.

Support for the previous LTS, Django 3.2, will end in April 2024. Then, [upstream support for PostgreSQL 11 ends in November 2023. Django 4.2 supports PostgreSQL 12 and higher.](https://docs.djangoproject.com/en/5.0/releases/4.2/#dropped-support-for-postgresql-11)

But [the EOL of PostgreSQL 12  is coming soon in November 14, 2024.](https://www.postgresql.org/support/versioning/)

It may be a lot of challenges.

NtskwK, 2024-08-19:
Hello everyone!

Great news! The migration work has made some progress, including Django 4.2 LTS, PostgreSQL 16, and more. We still need to conduct further testing on functions. Could you create a new branch for this purpose, please. I will creat a PR and advance the work to the next stage.

Thank you! @Saijin-Naib @smathermather

smathermather, 2024-08-19:
I defer to Piero on workflow, but we typically just keep the main branch except for hot fixes, and then merge from branches on forks when testing is complete. If you let me know what fork/branch you have for testing, I can build and test as a start.

NtskwK, 2024-08-19:
I have solved a part compatibility problems to sure it can work in the development environment.

But there are also many different from [django-rest-framework-jwt](https://github.com/jpadilla/django-rest-framework-jwt) to [djangorestframework-simplejwt](https://github.com/jazzband/djangorestframework-simplejwt). So it can not pass the test.

But I don't really understand about auth function. I'm sorry.

NtskwK, 2024-08-19:
> That is fantastic news! Thanks for continuing on this path.
> 
> I defer to Piero on workflow, but we typically just keep the main branch except for hot fixes, and then merge from branches on forks when testing is complete. If you let me know what fork/branch you have for testing, I can build and test as a start.

My apologies. I noticed that there is a actions for Build Docker Image and Public. I don't know whether it can bring dangers to other users with some **Breaking change**. In my opinions that we should be more careful. 

If you think it doesn't matter, that I will do it soon.

smathermather, 2024-08-20:
Thanks for the pull request!

We can run workflows on pull requests without merging, they just don't run by default. Once the pull request is far enough along and reviewed, I would expect we'll also test it in a workflow before merging.

NtskwK, 2024-08-25:
#1543 @pierotofy 
There is a problem with the recover database migrations.

> django.db.utils.ProgrammingError: type "raster" does not exist
> LINE 1: ...d_control_points" varchar(100) NULL, "orthophoto" raster NUL...

I found the field orthophoto in models.Task was been add in 2016, then deleted in 2017. 

>
```python
 79 |                 ('orthophoto', django.contrib.gis.db.models.RasterField(blank=True, help_text='Orthophoto created by OpenDroneMap', null=True, srid=4326)),
```

>app\migrations\0004_auto_20170707_1014.py
```python
    operations = [
        migrations.RemoveField(
            model_name='task',
            name='orthophoto',
        ),
```

`python manage.py migrate` will perform all the operations in turn. It could have to remove database migrations files if we don't change `app\migrations\0001_initial.py` with create `postgis_raster` extension.
[If you plan to use PostGIS raster functionality, you should also activate the postgis_raster extension. ](https://docs.djangoproject.com/en/5.1/ref/contrib/gis/install/postgis/#creating-a-spatial-database)

NtskwK, 2024-08-26:
TODO: update Ubuntu to 24.04 LTS or 22.04 LTS

I'll have a try.

NtskwK, 2024-08-26:
> sed: can't read /usr/include/proj_api.h: No such file or directory

The behavior of `libproj-dev ` did not match expectations on 24.04 even 22.04. What is the purpose of this command? Is this necessary? @pierotofy @chris-bateman #1545 

```dockerfile
	tar -zxf postgis-$POSTGIS_VERSION.tar.gz; \
	sed -i 's/#error.*/#define ACCEPT_USE_OF_DEPRECATED_PROJ_API_H 1/' /usr/include/proj_api.h; \
        cd /staging/postgis-$POSTGIS_VERSION; \
	./configure --with-pgconfig=/usr/local/pgsql/bin/pg_config --with-gdalconfig=/usr/bin/gdal-config > /dev/null; \
```

NtskwK, 2024-09-01:
```
worker      | [2024-09-01 12:31:09,435: ERROR/ForkPoolWorker-13] worker.tasks.process_task[7d945936-a740-43db-9ec3-1f65338a9cb2]: Uncaught error! This is potentially bad. Please report it 
to http://github.com/OpenDroneMap/WebODM/issues: Signal.__init__() got an unexpected keyword argument 'providing_args' Traceback (most recent call last):
worker      |   File "/webodm/worker/tasks.py", line 143, in process_task
worker      |     task.process()
worker      |   File "/webodm/app/models/task.py", line 629, in process
worker      |     resized_images = self.resize_images()
worker      |   File "/webodm/app/models/task.py", line 1154, in resize_images
worker      |     from app.plugins import signals as plugin_signals
worker      |   File "/webodm/app/plugins/signals.py", line 3, in <module>
worker      |     task_completed = django.dispatch.Signal(providing_args=["task_id"])
worker      | TypeError: Signal.__init__() got an unexpected keyword argument 'providing_args'
```

NtskwK, 2024-09-08:
@pierotofy  Django 4.2 only supports PostgreSQL 12 and higher. Whether to need to continue to push the work, but more conservative.